realsenseai/librealsense #4857: Windows Camera Frame Server memory leak

g2-bernotas, 2019-09-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   |  05.11.06.250  | 
| Operating System & Version |   Win 10  | 
| Platform                                 |  PC  |
| SDK Version                            |  2.25.0  | 
| Language                            |  C++                       | 
| Segment			|  others                           | 

### Issue Description
I am recording data using pause/resume as it was answered [here](https://github.com/IntelRealSense/librealsense/issues/4633). I have noticed that I may have some sort of a memory leak that piles up under the name of Windows Camera Frame Server:

[image]

I have seen that people have experienced this [issue](https://answers.microsoft.com/en-us/windows/forum/all/windows-camera-frame-server-memory-leak-leading-to/0894f547-f79d-4d64-b9ab-c3220b46a282#targetText=Windows%20Camera%20Frame%20Server%20%2D%20memory,called%20Windows%20Camera%20Frame%20Server.&targetText=All%20you%20have%20to%20do,back%20camera%20a%20few%20times.) with Skype as it constantly adapts the video quality. However, I haven't seen anyone have it for the RealSense camera. I am consistently starting and stopping the camera to record data (bagfiles). As seen from the image, there is almost 2GB memory allocated to the Frame Server after about 18 hours of recording (approx. 600 bagfiles were generated (up to 10s each)). Stopping the recording software and closing the camera has led to memory release, but I prefer not to stop/close it.

I tend to believe that this is a Windows issue, but I hope someone has had this problem and managed to solve it.

dorodnic, 2019-09-25:
Hi @g2-bernotas 
Thank you for the report. Windows Camera Frame Server service can be safely disabled AFAIK, and it is not critical to camera operation. 
We also provide secondary Windows driver (intended for Windows 7 but can be also installed on Win 10), but it is less convenient (will not work with non librealsense apps)

g2-bernotas, 2019-09-26:
Hi @dorodnic 

I am afraid Windows Camera Frame Server Service (WCFSS) is required to operate RealSense camera. I tried to end the WCFSS via Task Manager, which ended up in crashing the software (both mine and the RealSense Viewer), but after I started the software again, the WCFSS would restart, but everything would work (with increasing RAM once I am recording). The next step was to disable it as explained [here](https://www.winhelponline.com/blog/webcam-anniversary-update-windows-10-yuys-standard/); however, this also did not work, which led up to this error: 
[image]

Any suggestions?

RealSenseCustomerSupport, 2019-10-16:
----------------------------------------------

@g2-bernotas, were you able to workaround it and launch RealSense camera without camera frame server?
Please let us know if you have further question. Thank you.

g2-bernotas, 2019-10-16:
@RealSenseCustomerSupport as I mentioned in my last response, disabling the Windows Camera Frame Server Service did not allow to start the RealSense camera. I continue recording data, but I have to restart the Windows Camera Frame Server Service every few days to release the memory.

RealSenseCustomerSupport, 2019-10-22:
----------------------------------------------

Understood and I am glad to know that you can continue the recording with RealSense camears by restarting the Windows camera frame server service.
Just for reference, on my system (Win10 version 1809 build 17763), it was able to launch Realsense Viewer and stream RealSense camera without the WCFSS.
Please let us know if you have any further question.

closed 2019-10-31 by RealSenseCustomerSupport