realsenseai/librealsense #4858: librealsense example code running issue

manukharshan, 2019-09-13:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

liuyanqi, 2019-09-19:
I have the same issue

RealSenseCustomerSupport, 2019-10-18:
----------------------------------------------

Hi @manukharshan,

Please provide us with the following details about your system configuration :
how many cameras are in active use in the system
for each camera please provide :
    camera model (D435,D435i,etc)
    FW version
build number of RealSense SDK 2.0

Thank you!

RealSenseCustomerSupport, 2019-10-21:
----------------------------------------------

Hi @manukharshan,

Do you still need help with this question?

Thank you!

liuyanqi, 2019-10-21:
Hi.
The camera model I use is D415.

RealSenseCustomerSupport, 2019-10-25:
----------------------------------------------

Hi @liuyanqi,

Please provide more details about your system configuration :
how many cameras are in active use in the system? For each camera please provide FW version and
camera model. What build number of RealSense SDK 2.0 is installed on your system?
What operating system and platform do you experience the issue on?
How the camera(s) is/are connected to the host system?
Along with that information, please describe the exact scenario that leads to the problem.

Thank you!

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

@liuyanqi,

Do you still need assistance with this matter?

closed 2019-11-08 by RealSenseCustomerSupport