realsenseai/librealsense #4859: 24 FPS support for filmmaking use case

sam598, 2019-09-14:
It looks like Intel is now interested in promoting filmmaking as a use case for the RealSense cameras:
https://support.intelrealsense.com/hc/en-us/community/posts/360035016413-RealSense-technology-and-the-Ncam-Mk2-Camera-Bar

Will they provide support for it but adding industry standard framerates like 24 FPS?

Previously they said it was possible to add through firmware, but it would be too confusing for developers to add another enumeration to the SDK.
https://github.com/IntelRealSense/librealsense/issues/2865

If they are now promoting this they should seriously reconsider.

RealSenseCustomerSupport, 2019-11-06:
----------------------------------------------

Hi @sam59

People have been talking about 24 fps support.
There are actually several frame rates: 24 FPS, 23.98 FPS and 25 FPS.
Could you please elaborate what is the frame rate you are looking for and what is your application?
We are gathering the feedback and scoping the effort.

Thanks,

sam598, 2019-11-06:
23.98 is the most common framerate used for film, television and streaming production right now.

24.00 is an older standard that comes from celluloid film cameras. 25.00 is a standard framerate sometimes used in Europe and Asia.

23.98 would be my recommendation as it would get the most use from filmmakers, and allow the RealSense cameras to be used in film and television production.

As for my own use case, I am developing a small light field camera using an array of RealSense devices.

[image]

RealSenseCustomerSupport, 2019-11-07:
----------------------------------------------

Hi @sam59
Thanks for the great feedback!

RealSenseCustomerSupport, 2020-03-10:
----------------------------------------------

We will implement this, but the schedule is not available at this moment yet.

sam598, 2020-03-11:
@RealSenseCustomerSupport Thank you for the update, very exciting news!

If you need someone to beta test the firmware I would be very interested., I would be able to test it with a multicamera RealSense setup, and I am also ready to test syncronization with external 3rd party cinema cameras.

RealSenseCustomerSupport, 2020-04-27:
----------------------------------------------

Hi @Hi @sam59 Could you please confirm the 24 fps request is for depth camera, right? Not RGB camera.

sam598, 2020-04-27:
The priority is the depth camera. Is there an issue getting the RGB camera to run at 23.98fps as well?

If it is not possible to get the RGB camera to run in sync, the depth camera can still be synchronized with an external camera running at 23.98.

However having the RGB color run at the same framerate would greatly increase the usability of the device, and reduce the complexity and cost for end users.

agrunnet, 2020-06-10:
@sam59 we just released a new feature we call Genlock. It allows you to make our cameras slave to almost any external signal, so you can get your 23.98fps for example. Please check it out and see if it will work for you. 
https://www.intelrealsense.com/depth-camera-external-sync-trigger/

RealSenseCustomerSupport, 2020-07-07:
----------------------------------------------

@sam59 Do you have a chance to try out the Genlock? Is there any issue?

RealSenseCustomerSupport, 2020-07-15:
----------------------------------------------

@sam59 Do you have a chance to try out the new Genlock feature?

closed 2020-08-20 by dorodnic