realsenseai/librealsense #4860: I cannot updating Intel® RealSense™ D400 product family firmware mannually or automatically

liuyifei13, 2019-09-16:
| | |
|---|---|
|**librealsense**|2.25.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D4xx Recovery|
|**Product Line**|D400|
|**Asic Serial Number**|933623023875|

Please provide a description of the problem

liuyifei13, 2019-09-16:
when I connected my D435 camera to my laptop, there is something wrong with the viewer saying
intel RealSense D4xx Recovery Disconnected

RealSenseCustomerSupport, 2019-10-08:
----------------------------------------------

Hi

Sorry to hear about.
1) What is current firmware version on your D435? please run DFU Tool, and then

- select option "1, Update Camera Device.". What do you see from there? can you share the screenshot with any errors?

2) Maybe the calibration table is corrupted during firmware update. Please reset it by using Dynamic Calibration Tool. (https://downloadcenter.intel.com/download/28517/Intel-RealSense-D400-Series-Calibration-Tools-and-API).

- after installing the tool, go to the bin folder: e.g. "C:\CalibrationToolAPI\2.8.0.0\bin"

- Open Powershell window, and run the command: "**.\Intel.Realsense.CustomRW.exe -g**" (see attached pic).

3) Try update firmware again. If you still have the problem, please let us know.

Thanks and good luck.

Attachment(s):
GoldenTable_recovery.PNG - https://realsensesupport.intel.com/attachments/token/WLzzKlJ74xfGPIfuhxMdEwEcw/?name=GoldenTable_recovery.PNG

RealSenseCustomerSupport, 2019-10-08:
----------------------------------------------

Is there any more question about this issue?

Thanks

closed 2019-10-17 by RealSenseCustomerSupport