realsenseai/librealsense #4861: Cannot enumerate devices in python.

leaf918, 2019-09-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win7 | 
| Kernel Version (Linux Only)    | --                                        | 
| Platform                                 | PC  |
| SDK Version                            | 2                     | 
| Language                            |  python}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
with the following code to enumerate devices:
'context.devices'
It will enumerate one device twice ,while missing the other one.
There are 2 rs on my PC.

RealSenseCustomerSupport, 2019-10-07:
----------------------------------------------

Hi @leaf918,

Multiple cameras are likely not to work on Windows 7 at the moment. We are constantly expanding the list of supported platforms and features and will notify our customers and the community once this feature will become available for Windows 7.

Thank you!

RealSenseCustomerSupport, 2019-10-22:
----------------------------------------------

Hi @leaf918,

Do you need any further clarifications regarding this topic?

leaf918, 2019-10-23:
> Hi @leaf918,
> 
> Do you need any further clarifications regarding this topic?

No,and thanks for the explaination for this issue.
Regards

closed 2019-10-23 by leaf918