realsenseai/librealsense #4862: Pyrealsense2   How to Adjust temporal filter parameter

lalujesus, 2019-09-16:
| Camera Model                       |  D400 } | 
| Firmware Version                   |5.11.11.100 | 
| Operating System & Version |   Linux (Ubuntu 16/17) / MacOS  | 
| Kernel Version (Linux Only)    |   4.15                               | 
| Platform                                 | PC |
| SDK Version                            |  2.28.0                          | 
| Language                            | python                       | 
| Segment			|  {others }                          | 

[image]
[image]

When tried to estimate the real XYZ coordinates of of corner points of a detected object ( using SIFT algorithm for feature identification) and  plotted XYZ data  of corner points , It is not forming rectangle. It continuously oscillating.  Is it due to unstable depth frame. I applied temporal filter again not improving. If the problem is with temporal filter how to change l filter parameters like alpha , Delta  in python

lramati, 2019-09-18:
same way as in C++. You can either pass these parameters into the constructor for the temporal filter, or you can use the set_option API

lalujesus, 2019-09-21:
Thank you for the answer .  I will try

closed 2019-09-21 by lalujesus