realsenseai/librealsense #4865: has ir pattern

Johance2, 2019-09-16:
| | |
|---|---|
|**librealsense**|2.25.0 RELEASE|
|**OS**|Windows|
D415
Please provide a description of the problem
has white ir 
when i click install button on the right-top panel in Depth.Quality.Tool.exe
[image]

dorodnic, 2019-09-16:
Remove IR pattern preset only has effect when configuring Infrared stream to RGB8 format. Y8 will still produce raw infrared data. 
Also, even in RGB8 mode, the device is trying to remove the pattern using color correction (basically erasing the specific color of the projector) but it will not succeed 100%. This is one of the reasons for an additional RGB sensor.

RealSenseCustomerSupport, 2019-10-08:
----------------------------------------------

Is there any more question about this issue?

Thanks

closed 2019-10-11 by RealSenseCustomerSupport