realsenseai/librealsense #4868: How to check and change the frame rate of the D435i camera in MATLAB

Ahad19931, 2019-09-16:
| Camera Model                       | {D400 } | 
| Operating System & Version |   {Win (8.1/10)                                         | 
| Platform                                 | PC
| Language                            |  {Matlab }                          | 
| Segment			|  {Robot }                          | 

I am streaming the depth frame of the d435i camera on matlab but there seems to be a certain delay in each frame being streamed. Therefore, I was wondering if changing the fps rate may help in eliminating the delay. For this reason, I want to know how can I change the fps of the camera in MATLAB. Below is the code that I am using.
function align_serial_mix_2()
% Make Pipeline object to manage streaming
pipe = realsense.pipeline();
% Start streaming on an arbitrary camera with default settings
profile = pipe.start();
align_to = realsense.stream.color;
alignedFs = realsense.align(align_to);
% Get streaming device's name
dev = profile.get_device();
name = dev.get_info(realsense.camera_info.name);
% Get frames. We discard the first couple to allow
% the camera time to settle
delete(instrfind);
s=serial('COM9','Baudrate',57600,'Databits',8,'StopBits',1,'parity','none');
fopen(s);
while(true)
%for i = 0:1
fs = pipe.wait_for_frames();
%end
% Select depth frame
aligned_frames = alignedFs.process(fs);
depth = aligned_frames.get_depth_frame();
% get depth image parameters
depthSensor = dev.first('depth_sensor');
depthScale = depthSensor.get_depth_scale();
depthWidth = depth.get_width();
depthHeight = depth.get_height();
% retrieve UINT16 depth vector
depthVector = depth.get_data();
% reshape vector, and scale to depth in meters
depthMap = double(transpose(reshape(depthVector, [depthWidth,depthHeight]))) .* depthScale;
[r,c]=size(depthMap);
      for i=1:r
        for j=1:c
            if(depthMap(i,j)>=0.05 && depthMap(i,j)<=0.8)
                depthMap(i,j)=1;
        else
                depthMap(i,j)=0;
            end
        end
      end
%align the depth frames to the color stream
%Select rgb frame
color = fs.get_color_frame();
colordata=color.get_data();
img = permute(reshape(colordata',[3,color.get_width(),color.get_height()]),[3 2 1]);
%simplify binary image using morphological operations
img_fill=imfill(depthMap,'holes');
d=strel('disk',25);
imgM=imclose(img_fill,d);
img_area=50;
Iout1 = bwareaopen(imgM,500);
Iout=imresize(Iout1,0.3);
size(Iout)
[nrows ncols dim] = size(Iout);
% Get the slices colums wise split equally
left = Iout(:,1:floor(ncols/3),:);
middle = Iout(:,floor((ncols/3))+1:floor(2*ncols/3),:);
right = Iout(:,floor(2*ncols/3)+1:ncols,:);
%figure,
subplot(1,3,1), imshow(left,[]); title('left');
subplot(1,3,2), imshow(middle,[]); title('middle');
subplot(1,3,3), imshow(right,[]); title('right');
%imshow(Iout)
cc1=middle;
my_area1=struct2cell(regionprops(cc1,'Area'));
area1=cell2mat(my_area1)
cc2=left;
my_area2=struct2cell(regionprops(cc2,'Area'));
area2=cell2mat(my_area2)
cc3=right;
my_area3=struct2cell(regionprops(cc3,'Area'));
area3=cell2mat(my_area3)
%Start serial communication with robot
if(isempty(area1))
    pause(100/1000);
    val=hex2dec({'AA','0A','00','00','55'});
    fwrite(s,val);
else
    if(area1<=4000)
         pause(100/1000);
         val=hex2dec({'AA','0A','00','00','55'});
         fwrite(s,val);
    elseif(isempty(area2)|area2<=4000)
         %for a=0:1:15
             pause(100/1000);
             val=hex2dec({'AA','00','28','00','55'});
             fwrite(s,val);
         %end
    elseif(isempty(area3)|area3<=4000)
         %for a=0:1:15
             pause(100/1000);
             val=hex2dec({'AA','00','D8','00','55'});
             fwrite(s,val);
         %end
    else
         pause(100/1000);
         val=hex2dec({'AA','00','00','00','55'});
         fwrite(s,val);
         %for a=0:1:10
            pause(100/1000);
            val=hex2dec({'AA','F5','00','00','55'});
            fwrite(s,val);
         %end
    end
end
end
fclose(s);
% Stop streaming
pipe.stop();
end

RealSenseCustomerSupport, 2020-02-15:
----------------------------------------------

Please change the line `profile = pipe.start()` to

        cfg = realsense.config();
        %stream_type => 1: DEPTH, 2: COLOR, 3: IR
        %format => 1: Z16, 4: YUYV, 5: RGB8, 9: Y8, 14: UYVY
        %cfg.enable_stream(stream_type, stream_index, width, height, format, fps)
        cfg.enable_stream(1, -1, 640, 480, 1, 30) % for example
        profile = pipe.start(cfg);

RealSenseCustomerSupport, 2020-03-10:
----------------------------------------------

@Ahad19931 Did you get chance to try? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-03-20:
----------------------------------------------

Closing this ticket if there is no other question for this topic. Thank you.

closed 2020-03-20 by RealSenseCustomerSupport