realsenseai/librealsense #4869: DKMS build failing on 4.15.0-62

mikepurvis, 2019-09-16:
Was fine on 4.15.0-58, now is broken. See the build output below.

```
DKMS make.log for librealsense2-dkms-1.3.5 for kernel 4.15.0-62-generic (x86_64)
Mon Sep 16 18:37:27 UTC 2019
make: Entering directory '/usr/src/linux-headers-4.15.0-62-generic'
CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_driver.o
In file included from ./arch/x86/include/asm/atomic.h:5:0,
from ./include/linux/atomic.h:5,
from /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_driver.c:14:
./arch/x86/include/asm/qspinlock.h: In function ‘native_queued_spin_unlock’:
./arch/x86/include/asm/qspinlock.h:42:25: error: ‘struct qspinlock’ has no member named ‘locked’
smp_store_release(&lock->locked, 0);
^
./include/linux/compiler.h:304:9: note: in definition of macro ‘__compiletime_assert’
if (!(condition))     \
^
./include/linux/compiler.h:324:2: note: in expansion of macro ‘_compiletime_assert’
_compiletime_assert(condition, msg, __compiletime_assert_, __LINE__)
^
./include/linux/compiler.h:327:2: note: in expansion of macro ‘compiletime_assert’
compiletime_assert(__native_word(t),    \
^
./include/linux/compiler.h:327:21: note: in expansion of macro ‘__native_word’
compiletime_assert(__native_word(t),    \
^
./arch/x86/include/asm/barrier.h:97:2: note: in expansion of macro ‘compiletime_assert_atomic_type’
compiletime_assert_atomic_type(*p);    \
^
/var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/../../../../include-overrides/asm-generic/barrier.h:157:33: note: in expansion of macro ‘__smp_store_release’
#define smp_store_release(p, v) __smp_store_release(p, v)
^
./arch/x86/include/asm/qspinlock.h:42:2: note: in expansion of macro ‘smp_store_release’
smp_store_release(&lock->locked, 0);
^
```

ev-mp, 2019-09-18:
@mikepurvis  hello, we are looking into this as it similar to #4669 to resolve it for 4.15 branch.
In the meanwhile you can either try the proposed WA DKMS or run the manual patch script from the [installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md)
>./scripts/patch-realsense-ubuntu-lts.sh

mikepurvis, 2019-09-18:
@ev-mp I tried the deb posted in that ticket, and I get the same error with the `-62` kernel.

Are the scripts that build that package in this repo somewhere, or are they elsewhere?

ev-mp, 2019-09-19:
@mikepurvis ,  the link to run the [patches installation script](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) is available in the previous post.
The script itself and the relevant patches are maintained in the SDK's source code tree
Download/clone the source code and invoke the script from the SDK's root dir:
>./scripts/patch-realsense-ubuntu-lts.sh

mikepurvis, 2019-09-20:
Broken on `-64` as well, same error message in the build log.

iskwa, 2019-10-10:
FYI. I also got same error message on 4.15.0-65.

RealSenseCustomerSupport, 2019-10-10:
----------------------------------------------

We are currently working on a solution for this issue and update will be posted as soon as
a solution has been implemented.

Thank you!

mikepurvis, 2019-10-24:
@RealSenseCustomerSupport Status of this? I'm still seeing `1.3.5-0ubuntu1` as the most recent version of `librealsense2-dkms`.

ihadzic, 2019-10-27:
broken on 4.15.0-66 for me, same errors building

jaymzh, 2019-10-30:
@RealSenseCustomerSupport  - one week since last follow-up - any update?

kroq-gar78, 2019-11-09:
I am still getting the DKMS error on `librealsense2-dkms` version `1.3.6-0ubuntu0`, and `librealsense2` version `2.30.0-0~realsense0.1682`.

jaymzh, 2019-11-09:
@RealSense-Customer-Engineering / @RealSenseCustomerSupport  - we're at the 1-month mark since your last comment. Any update?

ev-mp, 2019-11-14:
@mikepurvis , @jaymzh , the fix version is available for download via PPA.
Please check and provide feedback

ihadzic, 2019-11-14:
I updated this morning and it pulled 1.3.7 which seems to have fixed the problem. It works for me now.

jaymzh, 2019-11-15:
Is the new release expected to work with both -58 and -64?

ihadzic, 2019-11-15:
> Is the new release expected to work with both (the older, previously working) -58 and (newer, previously broken) -64?

Just learned the hard way: the update fixed the build in newer kernel but broke the old kernel. That actually bites me hard, because I have two systems. One is based on generic ITX board and the other is based on UpBoard. The former tends to be ahead in the kernel than the latter. Of course, my UpBoard build is now broken with the update, because it's a 4.15.0-37 kernel. My ITX-board build works because it's 4.15.0-70 kernel.

Really, pretty please, can we get this fixed for real, not by fixing one, while breaking the other. The crux of the problem is that Intel maintains the kernel modules out-of-tree instead of properly working with the kernel development community to push this upstream. They do this well for other products (graphics chipset, for example), but Realsense is a mess.

ihadzic, 2019-11-15:
FWIW, I am able to get the system with an older kernel to work by downgrading librealsense2-dkms package to 1.3.5 and having everything else the latest:

```
apt update
apt upgrade       # realsense dkms will fail to build on -58 and older
# downgrade realsense dkms
apt install librealsense2-dkms=1.3.5-0ubuntu1
```

ev-mp, 2019-11-19:
@ihadzic , thanks for the feedback. I hope the latest v1.3.9/v1.3.10 versions bring this to the end as they underwent extra testing and very physically deployed on multiplicity of kernel for verification.
See #5249 for details.
Just to confirm - using dkms v1.3.5 with kernels prior to 4.15.0-58 is perfectly fine and there is no need for immediate upgrade, unless you're planning to upgrade Ubuntu to 4.15.0-58+.

ihadzic, 2019-11-22:
1.3.9 seems to build fine on all kernels that I am dealing with at the moment.

ev-mp, 2019-12-01:
@mikepurvis hello, did you manage to deploy and verify that version ?
Please update the status.

mikepurvis, 2019-12-03:
With v1.3.7 of `librealsense2-dkms`, I get all four modules successfully building for `4.15.0-47` and `4.15.0-72`. Thanks for your attention to this!

closed 2019-12-03 by mikepurvis