realsenseai/librealsense #4872: Cannot build opencv dnn example

catmasteryip, 2019-09-17:
| Camera Model                       |  D400  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu18  | 
| Kernel Version (Linux Only)    |>=5 | 
| Platform                                 | PC |
| SDK Version                            | 2.28 | 
| Language                            |  C#/opencv                     | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Cannot build opencv dnn example; I have installed opencv and checked there is dnn-related objects inside the opencv source folders, yet when building rs-dnn the opencv library cannot be detected
[image]
```
/home/ytf/librealsense/wrappers/opencv/dnn/rs-dnn.cpp:5:10: fatal error: opencv2/dnn.hpp: No such file or directory
 #include <opencv2/dnn.hpp>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
wrappers/opencv/dnn/CMakeFiles/rs-dnn.dir/build.make:62: recipe for target 'wrappers/opencv/dnn/CMakeFiles/rs-dnn.dir/rs-dnn.cpp.o' failed
make[2]: *** [wrappers/opencv/dnn/CMakeFiles/rs-dnn.dir/rs-dnn.cpp.o] Error 1
CMakeFiles/Makefile2:424: recipe for target 'wrappers/opencv/dnn/CMakeFiles/rs-dnn.dir/all' failed
make[1]: *** [wrappers/opencv/dnn/CMakeFiles/rs-dnn.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

ev-mp, 2019-09-18:
@catmasteryip , all the SDK examples were built with openCv 3.4.
What version are you running and can you verify that the highlighted header file `<opencv/dnn.hpp>` is present on hard-disk ?

When building from sources openCV was reported having issues with dnn when statically compiled.

RealSenseCustomerSupport, 2019-10-12:
----------------------------------------------

Hi @catmasteryip,

Did you have a chance to verify that the above-mentioned header file <.../dnn.hpp> is located correctly?

closed 2019-12-10 by dorodnic

Rafipaulino, 2021-05-04:
Hi, I'm tring to compile this project in a jetson nano with a intel realsense d345i camera and I'm getting the same error. Shuould I compile the wrappers examples with the complete SDK (as I did with the examples)?