realsenseai/librealsense #4873: Remove points before exporting to PLY

visus100, 2019-09-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | D400_Series_FW_5_11_11_100 | 
| Operating System & Version |   Win (8.1/10) | 
| Platform                                 | PC |
| SDK Version                            | SDK-WIN10-2.23.0.909                      | 
| Language                            |  C++                    | 

### remove area of points

I need to remove area of points straight along Y axis (simplest possible remove) just like in picture below (red rectangle) and after that export resault to PLY.

[image]

### Code RealSense Pointcloud Example

    int main(int argc, char * argv[]) try{
    window app(1280, 720, "RealSense Pointcloud Example");
    glfw_state app_state;
    register_glfw_callbacks(app, app_state);

    rs2::pointcloud pc;
    rs2::points points;

    rs2::pipeline pipe;
    pipe.start();

    int helper = 0;

    while (app) {
        // Wait for the next set of frames from the camera
        auto frames = pipe.wait_for_frames();
        auto color = frames.get_color_frame();
        pc.map_to(color);
        auto depth = frames.get_depth_frame();
        points = pc.calculate(depth);
        // Remove specyfic area of points
        
        app_state.tex.upload(color);
        draw_pointcloud(app.width(), app.height(), app_state, points);

		if (helper==100) {
			std::cout << "PLY Saved";
			points.export_to_ply("filename.ply", color);
		}
		helper++;
    }

    return EXIT_SUCCESS;
    }

### ---------

I couldn't find any example of removing specyfic area points only transform, rotate and measure example.
Any suggestions?
Thanks

wodndb, 2019-09-30:
You are right, there are no example about that.
You should write code to remove some points in raw point cloud data.

RealSenseCustomerSupport, 2019-10-17:
----------------------------------------------

Closing this ticket if there's no further question or help needed on this topic.
Thank you.

closed 2019-10-24 by RealSenseCustomerSupport