realsenseai/librealsense #4875: We are observing error messages as mentioned in the log file.We are using below code to get colour frames from RealSense camera and streaming the same over HTTP.But after sometime colour stream stucks after sometime.

harshars08, 2019-09-17:
[Realsense_logs.txt](https://github.com/IntelRealSense/librealsense/files/3621082/Realsense_logs.txt)
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

harshars08, 2019-09-17:
Code to get colour frames:

public static IEnumerable<System.Drawing.Image> realSenseColorFrames()
        {
            
            log.Info("StreamApi::realSenseColorFrames Getting RealsenseClorFrames!!");
            
            Align align = new Align(Intel.RealSense.Stream.Color);
            try
            {
                while (true)
                {
                    using (var frames = pipe.WaitForFrames(10000))
                    {
                        log.Info("StreamApi::WaitForFrames!!");
                       
                        var frames2 = align.Process(frames).DisposeWith(frames);
                        
                        
                        using (VideoFrame vf = frames2.ColorFrame)
                        {
                            Console.WriteLine("Inside bitmap");
                            log.Info("StreamApi::VideoFrame!!");
                            Bitmap bmpfile = FrameToBitmap(vf);
                            yield return bmpfile;
                        }
                    }
                }
               
            finally
            {
                log.Debug("StreamApi::realSenseColorFrames finally block called ");
                
            }

        }

RealSenseCustomerSupport, 2019-10-08:
----------------------------------------------

I'd recommend to upgrade to more recent SDK build and camera firmware to take advantage of many bug fixes and enhancements made since firmware 5.9.2.  The newer SDK build also includes sample c# apps (frame alignment) which can be used for testing if needed.
Thank you.

closed 2019-10-17 by RealSenseCustomerSupport