realsenseai/librealsense #4876: Velocity drift in T265 while robot is stationary

msadowski, 2019-09-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.1.0.279 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.0.0-27                                         | 
| Platform                                 | PC  |
| SDK Version                            |  latest used by the ROS driver                         | 
| Language                            |  C++ (ROS)                        | 
| Segment			|  Robot                        | 

### Issue Description
Hi!

I noticed twice already an issue when the velocity odometry from T265 starts drifting while the robot is stationary. So far I think I've only seen this behaviour when I set enable pose jumping to false.

Here is a gif showing the drift:
[image]

In the above gif you can see that realsense is reporting velocity of the robot forward and to the right. All this in an environment that I imagine should be quite feature rich. I'm feeding odometry data to realsense and I confirmed that the odometry data reports the robot stationary.

Here is how it looks like in raw odometry data (top - wheel odometry in x,y direction (ROS convention), bottom - odometry from T265):
[image]

And zoomed in to the region of interest:
[image]

Any ideas why this might be happening?

BriceRenaudeau, 2019-10-15:
I have the same problem and I figured out that the velocity drift appeared also while moving:
https://github.com/IntelRealSense/librealsense/issues/5004 (comparition between the wheel and  the visual odometry)

RealSenseCustomerSupport, 2019-11-19:
----------------------------------------------

@msadowski - Were you able to test the drift out with the latest librealsense 2.30.0 version?  I'd like to know if there has been any improvement in your scenario with this updated version.

Can you provide a video or link to a video showing the robot movement and the environment you're operating in?

msadowski, 2019-11-26:
@RealSenseCustomerSupport at the moment I don't have the Realsense set up properly on my robot, will try to re-enable it and run these tests soon!

RealSenseCustomerSupport, 2020-01-27:
----------------------------------------------

Hi @msadowski

I hope you were successful in running with the latest versions of librealsense.  Now we are up to version 2.32.1.  Let me know if further assistance is necessary or we can close this thread.

msadowski, 2020-01-28:
@RealSenseCustomerSupport Many thanks for remembering about this issue! It looks like I'll not be able to test it out for the next 2 months or so. If it's a clutter for you feel free to close it and I'll create a new issue if I run into it again.

RealSenseCustomerSupport, 2020-02-07:
----------------------------------------------

Hi @msadowski

Thanks for the update and do create a new one when you're back to working with the product.

closed 2020-02-07 by RealSenseCustomerSupport