realsenseai/librealsense #4877: System.AccessViolationException in .NET wrapper when getting the distance of a point

morcibacsi, 2019-09-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR305| 
| Firmware Version                   | 3.27.3.0 | 
| Operating System & Version |   Win 10 Pro 1903 | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC |
| SDK Version                            |  2.28.0.1085                          | 
| Language                            |  C#                          | 
| Segment			|  Object detection                          | 

### Issue Description

I am using the .NET wrapper and EmguCV to detect objects and their real world size. Sometimes I fail to detect the coordinates of the object properly. When this happens I pass invalid coordinates to the GetDistance(x, y) method of the DepthFrame class and the method raises a System.AccessViolationException. Of course I can avoid this with additional checks (like in the screenshot below), however maybe it would be nice if the method would fail in a more developer-friendly way (like a custom exception or something like that, or a return value which indicates the error).

[image]

ev-mp, 2019-09-18:
@morcibacsi , according to the attached screenshot you're passing a negative `y`  as pixel coordinate to the function, which is out of the [0-height] range
[image]

morcibacsi, 2019-09-18:
@ev-mp I am fully aware of that. As you can see on the screenshot, I have the necessary checks (it is just commented out) to avoid the Exception. My point was that the AccessViolationException is not the best to throw from the SDK. An ArgumentOutOfRangeException would be more appropriate.

ev-mp, 2019-09-18:
@morcibacsi , I agree that the input validation should be performed by SDK to improve the users' experience.
This will be fixed in the next releases

RealSenseCustomerSupport, 2020-01-30:
----------------------------------------------

Hi @morcibacsi, Do you have further questions on this? Can we close it?
Thanks,

morcibacsi, 2020-01-30:
@RealSenseCustomerSupport Can be closed. Thanks for the fix.

closed 2020-01-30 by morcibacsi