realsenseai/librealsense #4878: Issues translating Hello (and the rest) to Unity

CamAtlantic, 2019-09-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.0                          | 
| Language                            |  C# Unity                           | 
| Segment			| Installation art                     | 

Hi folks, I am having some issues with the Hello example when trying to translate to c# for Unity. Would love some help.

I have things set up such that I am able to print the distance correctly, however after 15-30 frames I start receiving the following error:
`ExternalException: rs2_pipeline_wait_for_frames(pipe:000000003E44A800)
Rethrow as Exception: Frame didn't arrived within 1000`
After that point, I do sometimes get frames coming through but it's very rare, and after a minute I'm up to hundreds of errors and usually less than 50 successful prints.

I understand what the error is saying (frame didn't show up so it timed out) but I don't understand why it's happening, nor why it consistently fails after 15-30 frames. The built example that comes with the SDK runs fine on my machine, and so does the RS Viewer.

Part of my problem is just the nature of trying to tease out the Unity version of the code without many examples - I'm not sure what parts of it are necessary and what everything is doing. I'm definitely getting there but I'm not confident moving more toward what I want to do without understanding why I'm losing 99% of frames.

Anyway, thanks for having a look at this and hope you have some advice!
Cheers,
Cam

RealSenseCustomerSupport, 2019-10-18:
----------------------------------------------

Hi @ CamAtlantic,

Please provide us with the following details about your system configuration :
how many cameras are in active use in the system
build number of RealSense SDK 2.0
build number of Windows 10

Along with that information, please describe the exact scenario that leads to the problem.

Thank you!

RealSenseCustomerSupport, 2019-10-22:
----------------------------------------------

Hi @ CamAtlantic,

Do you still need help with this question?

closed 2019-10-29 by RealSenseCustomerSupport