realsenseai/librealsense #4879: How to properly grab new frames on a timer tick?

brettmilinski, 2019-09-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Win(10) |                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.25 | 
| Language                            |  C#                  | 

### Issue Description
I am creating a Windows Service app that runs the T265 tracking camera and records info to a database. My code works for about the first 10-15 frames by properly recording location and direction (up/down, left/right, and forward/backward), but then it gets "stuck" on the same frame/position. It does not crash, but has the same value repeated over and over, so it seems like I'm simply reusing the same frame and the same frame data after a certain point.

I am using a timer tick event to retrieve the frames and record the information to my database. Is there any better way to approach this and guarantee that new frames will be picked up?

My code is below:

`public void OnTimer(object sender, ElapsedEventArgs args)`

       {          
            //get all the frame info
           try
           {
               using (var frames = pipelines[0].WaitForFrames())
           {

                    var f = frames.FirstOrDefault(Intel.RealSense.Stream.Pose);
                    var pose_data = f.As<PoseFrame>().PoseData;

                    if (lastX != (float)-300)
                    {
                        //x
                        if (lastX > pose_data.translation.x)
                        {
                            dx = "LEFT";
                        }
                        else if (lastX < pose_data.translation.x)
                        {
                            dx = "RIGHT";
                        }
                        else
                        {
                            dx = "";
                        }

                        //y
                        if (lastY > pose_data.translation.y)
                        {
                            dy = "DOWN";
                        }
                        else if (lastY < pose_data.translation.y)
                        {
                            dy = "UP";
                        }
                        else
                        {
                            dy = "";
                        }
                        //z
                        if (lastZ > pose_data.translation.z)
                        {
                            dz = "FORWARD";
                        }
                        else if (lastZ < pose_data.translation.z)
                        {
                            dz = "BACKWARD";
                        }
                        else
                        {
                            dz = "";
                        }
                    }
                    using (sw = new StreamWriter(path))
                    {
                        sw.WriteLine("Count: " + count.ToString() + " LastX: " + lastX + " LastY: " + lastY + " LastZ: " + lastZ + " X: " + pose_data.translation.x + " Y: " + pose_data.translation.y + " Z: " + pose_data.translation.z + " Moving: " + dx + " " + dy + " " + dz);
                        count++;
                    }
                    //Set last x,y,z to see what change occurred
                    lastX = pose_data.translation.x;
                    lastY = pose_data.translation.y;
                    lastZ = pose_data.translation.z;
                    dx = "";
                    dy = "";
                    dz = "";

                  
                }
            }
            catch(Exception e)
            {
                //using (sw = new StreamWriter(path, false))
                using (sw = new StreamWriter(path, false))
                {
                    sw.WriteLine(e.ToString());
                }
            }

            //update the DB based on the frame info
            try
            {
                using (SqlConnection connection = new SqlConnection())
                {
                    using (SqlCommand command = new SqlCommand())
                    {
                        connection.ConnectionString = connString;
                        connection.Open();
                        command.Connection = connection;
                        command.CommandText = "UPDATE Tracking SET [X]=@x, [Y]=@y, [Z]=@z, [Left]=@left, [Right]=@right, [Up]=@up, [Down]=@down, [Forward]=@forward, [Backward]=@backward";
                        command.Parameters.AddWithValue("@x", lastX);
                        command.Parameters.AddWithValue("@y", lastY);
                        command.Parameters.AddWithValue("@z", lastZ);
                        //LEFT OR RIGHT?
                        if (dx.Equals("LEFT"))
                        {
                            command.Parameters.AddWithValue("@left", 1);
                            command.Parameters.AddWithValue("@right", 0);
                        }
                        else if(dx.Equals("RIGHT"))
                        {
                            command.Parameters.AddWithValue("@left", 0);
                            command.Parameters.AddWithValue("@right", 1);
                        }
                        else
                        {
                            //neither
                            command.Parameters.AddWithValue("@left", 0);
                            command.Parameters.AddWithValue("@right", 0);
                        }
                        //UP OR DOWN?
                        if(dy.Equals("UP"))
                        {
                            command.Parameters.AddWithValue("@up", 1);
                            command.Parameters.AddWithValue("@down", 0);
                        }
                        else if(dy.Equals("DOWN"))
                        {
                            command.Parameters.AddWithValue("@up", 0);
                            command.Parameters.AddWithValue("@down", 1);
                        }
                        else
                        {
                            //neither
                            command.Parameters.AddWithValue("@up", 0);
                            command.Parameters.AddWithValue("@down", 0);
                        }
                        //FORWARD OR BACKWARD?
                        if(dz.Equals("FORWARD"))
                        {
                            command.Parameters.AddWithValue("@forward", 1);
                            command.Parameters.AddWithValue("@backward", 0);
                        }
                        else if(dz.Equals("BACKWARD"))
                        {
                            command.Parameters.AddWithValue("@forward", 0);
                            command.Parameters.AddWithValue("@backward", 1);
                        }
                        else
                        {
                            //neither
                            command.Parameters.AddWithValue("@forward", 0);
                            command.Parameters.AddWithValue("@backward", 0);
                        }
                        //commit updates to DB
                        command.ExecuteNonQuery();
                    }
                    connection.Close();
                }
            }
            catch (Exception e)
            {
                //log the exception               
                using (sw = new StreamWriter(path, false))
                {
                    sw.WriteLine(e.Message);
                }
            }
        }

morcibacsi, 2019-09-18:
I would try a different aproach. I would create a thread where I would run something like this:

```
var cfg = new Config();
cfg.EnableStream(Stream.Depth, 640, 480);
cfg.EnableStream(Stream.Color, Format.Rgb8);
var pp = pipeline.Start(cfg);

using (var releaser = new FramesReleaser())
{
    using (var frames = pipeline.WaitForFrames().DisposeWith(releaser))
    {
        if (frames.DepthFrame == null || frames.ColorFrame == null)
        {
            continue;
        }

        _frameProcessorMethod(frames, releaser);
    }
}

```
where the frameProcessorMethod would be an action, injected at the constructor:
```
private readonly Action<Frame, ICompositeDisposable> _frameProcessorMethod;
```
This way you can decouple the db logic from the camera processing code.

Or another thing which worth a shot is to put the whole camera initialization, and database stuff into your timer, and at the end of the method release the camera.

brettmilinski, 2019-09-19:
@morcibacsi Hi, thank you for your suggestions. I will definitely try them. 

I have one question about them though, how do I inject the action at the constructor?

I have experience with threading so that makes perfect sense, but I am quite a newbie when it comes to working with these cameras. That being said, any explanation on how to implement the suggested frameProcessorMethod would be appreciated.

RealSenseCustomerSupport, 2019-10-17:
----------------------------------------------

Hi Brett. Do you have everything you need on the original issue you were having here?

closed 2019-10-25 by RealSenseCustomerSupport