realsenseai/librealsense #4880: Pyrealsense2 BUS ERROR when doing pipeline.start()

EnrikDS, 2019-09-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435i } | 
| Firmware Version                   | (5.11.11.100 and before) | 
| Operating System & Version |   {Raspbian Buster | 
| Kernel Version (Linux Only)    |  (4.19.66-v7 +armv7l                                         | 
| Platform                                 | Pi3B+ |
| SDK Version                            |  {2.2 }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot/ }                          | 

### Issue Description
Hi,
after days of working I managed to compile the realsense for Raspberry Pi Buster and some Open3D libraries I am using. 
I updated to the latest firmware although these problems were also happening with previous one. 
I am able to start the interface realsense-viewer and it works but with some issues:
-Stereo module works fine most of the times. Although sometimes is breaks.
-RGB works fine util you turn it off, and then it gives you a sementation fault, and exits.
-The motion module does not work. Program stops and the output in the shell says:
**_17/09 17:04:34,733 WARNING [1808110656] (messenger-libusb.cpp:94) bulk_transfer returned error, endpoint: 134, error: Resource temporarily unavailable, number: 11
 17/09 17:04:34,744 WARNING [1808110656] (messenger-libusb.cpp:94) bulk_transfer returned error, endpoint: 134, error: Resource temporarily unavailable, number: 11
Bus error_**

However, my main problem is that I cannot use pyrealsense2 to get images form python. If I want to run the following script:
# First import the library
import pyrealsense2 as rs
pipeline = rs.pipeline()
pipeline.start()

As soon as I execute pipeline.start() I get a bus error, and the python program exits.
I have the intuition that the USB can only cope with certain transmissions and this should be configured properly before starting the pipeline. However, this is just an intuition and I do not know what I am talking about.

I would appreciate any help on this please. 

Many thanks

g2-bernotas, 2019-09-18:
Following your intuition, can you try to start with a specific config, which is set to the lowest resolution?

``` Python
import pyrealsense2 as rs

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 424, 240, rs.format.z16, 6)
pipeline.start(config)
```

EnrikDS, 2019-09-18:
Hi, many thanks for the idea. I did not know how to do that.
It turns out that it does not work, this time a different error when trying to start the pipeline with the desired configuration:

It says: RuntimeError: Couldn't resolve requests
just after the execution of pipeline.start(config)

Thanks

EnrikDS, 2019-09-18:
Ok, that gets solved with good configuration like:
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)

Although it gives a different error now, I think there is  problem with the compilation of pyrealsense2. 

Traceback (most recent call last):
  File "camera_test.py", line 9, in <module>
    import pybackend2 as rs
ImportError: /usr/local/lib/pybackend2.cpython-37m-arm-linux-gnueabihf.so: undefined symbol: _ZN12librealsense8platform14create_backendEv

Thanks

EnrikDS, 2019-09-18:
Sorry that gets solved with the correct configuration of the camera:
config.enable_stream(rs.stream.depth, 424, 240, rs.format.z16,6)

g2-bernotas, 2019-09-18:
So at what stage are you right now? I have never worked with the RealSense and Raspberry Pi. Have you followed any tutorials for setting up the RPi, such as this [one](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md?source=post_page-----882933271f3d----------------------)?

EnrikDS, 2019-09-18:
Everything seems to be working now. I followed that tutorial yes, but I found many problems when compiling realsense for Buster. And also, I am using Open3D as well, and run into trouble compiling. 
In order to get realsense working with buster I had to follow these steps:

**_When installing protobuff with python 3.7 there are some modifications to be done beforehand as the C wrappers do not compile. 
Look on Add python 3.7 compatibility (#4862) in github, protocolbuffers/protobuf

Additional modifications to the CMakeLists.txt as described in:
https://github.com/IntelRealSense/librealsense/issues/4375
To include the flag -latomic in most places._**

Hopefully this will help other people.

rofreli, 2019-09-23:
I have almost same configuration, just the raspberry version is different I use pi 4, but with me, Motion data (accel and gyro) doesn't work [#4899 ](https://github.com/IntelRealSense/librealsense/issues/4899#issuecomment-534068446). When I run realsense-viewer I got "Error type: No data to return (-21)" and when I try to execute Motion Module realsense-viewer close with BUS Error. Did you execute something diferent to install the librealsense in your raspbian?

EnrikDS, 2019-09-23:
Both the accelerometer and gyro do not work for me in realsense-viewer. They also do not work with the applications rs-pose or rs-motion. 
If I enable the motion module in realsense-viewer, the application switches-off and the message displayed in the terminal is WARNING [1736950848] (messenger-libsub.cpp:94) bulk transfer returned error, endpoint:134, error: Resource temporarily unavailable, number:11
Bus error

It is probably that usb is unable to handle that. Unfortunatelly I cannot lower the frequency more than that. Although the Raspberry pi 4 should not have these problems with USB3 I guess.

rofreli, 2019-09-23:
I had the same problem with USB3 and pi 4. 
I tried some workarounds but nothing works.

Thx

nizqsut, 2019-10-16:
> Following your intuition, can you try to start with a specific config, which is set to the lowest resolution?
> 
> ```python
> import pyrealsense2 as rs
> 
> pipeline = rs.pipeline()
> config = rs.config()
> config.enable_stream(rs.stream.depth, 424, 240, rs.format.z16, 6)
> pipeline.start(config)
> ```

I meet the same error when I run rs-pose. 
 I had tried run your code, got the as result at notebook:
```
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-1-ec7f5d9610d3> in <module>
      4 config = rs.config()
      5 config.enable_stream(rs.stream.depth, 424, 240, rs.format.z16, 6)
----> 6 pipeline.start(config)

RuntimeError: Couldn't resolve requests
```

What's the real problem?

EnrikDS, 2019-10-25:
Sorry, I do not have access to the camera anymore, maybe in the next months.

RealSenseCustomerSupport, 2019-11-07:
----------------------------------------------

@EnrikDS I think you're using USB2 connection on Raspberry Pi B. For USB2 mode, 424x240@6fps is not supported. So please try below supported configuration.

1280x720@6fps

640x480@6,15,30fps

480x270@6,15,30,60fps

RealSenseCustomerSupport, 2019-12-02:
----------------------------------------------

@EnrikDS Will close it if no other questions. Thanks!

closed 2019-12-02 by RealSenseCustomerSupport