realsenseai/librealsense #4881: Realsense Viewer won't open in Ubuntu 18.04

alextac98, 2019-09-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | N/A | 
| Firmware Version                   | ??? | 
| Operating System & Version |  Ubuntu 18.04  | 
| Platform                                 | PC |
| SDK Version                            |  2.28                         | 
| Language                            | N/A                       | 
| Segment			|  N/A    | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Every time I try to open the `realsense-viewer`, I get the following error: 
``` bash
12:30:13.808 [13923] [E] infra/Poller: poll: epoll_wait error 4
12:30:13.808 [13923] [E] Dispatcher: handleEvents(): Poller::poll() ret -1
 17/09 12:30:13,937 WARNING [140086725306112] (backend-v4l2.cpp:1148) Pixel format 20303157-0000-0010-8000-00aa003 likely requires patch for fourcc code W10 !
 17/09 12:30:13,951 WARNING [140086725306112] (backend-v4l2.cpp:1148) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 17/09 12:30:14,105 ERROR [140086725306112] (types.h:304) xioctl(VIDIOC_G_CTRL) failed Last Error: Invalid argument
 17/09 12:30:14,107 WARNING [140086725306112] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Enable / disable auto-exposure
 17/09 12:30:23,069 WARNING [140086331045632] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
```
Sometimes, it crashes immediately, and sometimes it crashes after I try to view the image coming from my cameras. The cameras aren't problematic because they work on other computers just fine. Any help is greatly appreciated!

ev-mp, 2019-09-18:
@alextac98 , the errors are due to communication/hw issues between host and the device.
The second error often indicates that the device was forcefully disconnected from host due to insufficient power or a similar issue.
>get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device

Knowing the camera type/fw version and also attaching the kernel logs when those errors occur is needed for review.
You haven't specified the platform you're using that produces those errors, and whether the setup/cabling is the same as with the other PCs where the camera work correctly. But in many cases switching to an external self-powered USB3 hub was able to mitigate power-related issues (if that proves to be correct)

g2-bernotas, 2019-09-18:
I only know from Windows side, but I had something similar when the SDK crashes every time you want to change any parameter (exposure, laser etc etc.). My solution was reported [here](https://github.com/IntelRealSense/librealsense/issues/4784).

alextac98, 2019-09-18:
@ev-mp It seems like the hardware was resetting due to power limitations by my laptop (I was plugging it directly into my laptop port). Now that I have plugged the cameras into a powered port, it works as expected. Thank you for your help!

closed 2019-09-18 by alextac98