realsenseai/librealsense #4883: rs-ar-basic object not stationary

JBBee, 2019-09-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.1.0.279 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC |
| SDK Version                            | 2.28.0                     | 
| Language                            |  C++                          | 
| Segment			| AR                         | 

### Issue Description
We're trying to solve an issue in our AR application that uses the T265 where an AR object doesn't stay 'in place' in reference to a real-world object when viewing it from different perspectives.  Obviously this could be due to a lot of factors in our own application, however, we're seeing the same issue in rs-ar-basic.exe.  In rs-ar-basic I place the 3 lines just above a real world object and it looks ok from the starting perspective.  Then I walk around the room and view it from another perspective it appears to have moved off of the real-world object I placed it on.  And then when I return to the starting perspective it is back on the real-world object. 

Below is a video of this occurring:  If you wouldn't mind taking a look you'll see after I walk around the table and view the lines they're no longer on top of the gray square, but then when I return to the starting perspective it's back on the square. 

Can you provide any insight as to why this is occurring?

Thanks.

[rs-ar-basic video](https://youtu.be/ILCRj_cltj8)

radfordi, 2019-09-18:
Thanks for the video, @JBBee.   At first glance, this looks like accumulated loop error.  Does it happen on a second loop (such that we have presumably already closed the loop internally and know you are coming around to the object)?

JBBee, 2019-09-18:
Hi @radfordi 

Thanks for taking a look.  This time I tried looping the table 3 times but it still appears off the square each time around when looking at it from the other perspective.  

Below is a video of the loops, but also at the 1 minute mark I just tried moving from the first perspective to the second perspective, without going around the table, and you can see the 3 lines slide off the square as I move around it to the second perspective.  And then when I move back to the first perspective it slides back on the square.  

Thanks again for your help.

[rs-ar-basic video 2](https://youtu.be/SpsVKw6r5WY)

JBBee, 2019-09-26:
Hi @radfordi,

I noticed that the 3 lines were off a distance similar to the extrinsics from the pose stream to the fisheye stream, so I tried reversing [this line](https://github.com/IntelRealSense/librealsense/blob/81d469db173dd682d3bada9bd7c7570db0f7cf76/examples/ar-basic/rs-ar-basic.cpp#L74), so it's now:
`rs2_extrinsics extrinsics = pipe_profile.get_stream(RS2_STREAM_POSE).get_extrinsics_to(fisheye_stream);`

After this change the 3 lines stay in place much better.  What do you think of this change?

Thanks.

radfordi, 2019-09-26:
Nice find, @JBBee!  See #4945.  Would you mind posting a follow up video?

JBBee, 2019-09-27:
Hi @radfordi,

Thanks.  Below is a video of it staying in place really well with the patch.  On other runs it has jumped around a little but this is great overall.  

[rs-ar-basic video 3](https://youtu.be/4aHwM9spR2Y)

RealSenseCustomerSupport, 2019-10-23:
----------------------------------------------

Thanks @JBBee for including the updated video and the code change for the sample.  Is there anything else on this thread need?

JBBee, 2019-10-23:
Nope, @RealSenseCustomerSupport .  I just wasn't sure if I should close it or you should.  I'll close it now though.  Thanks.

closed 2019-10-23 by JBBee