realsenseai/librealsense #4884: Unable to build nodejs wrapper (windows 10)

Marenix, 2019-09-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   || 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |   | 
| Platform                                 | PC |
| SDK Version                            |  2.20.0                         | 
| Language                            |  nodejs                         | 
| Segment			|                         | 

I've been trying to build node.js wrapper for librealsense2 (from source), but I keep getting an error and am unable to. I don't know whether I made a mistake or something somewhere, but I followed the instructions and I don't see where I could've made it.

```20>------ Build started: Project: BUILD_NODE_ADDON, Configuration: Debug Win32 ------
20>Generating ../../../wrappers/nodejs/build/Release/node_librealsense.node
20>'env' is not recognized as an internal or external command,
20>operable program or batch file.
20>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(209,5): error MSB6006: "cmd.exe" exited with code 9009.
20>Done building project "BUILD_NODE_ADDON.vcxproj" -- FAILED.
```

closed 2019-09-27 by Marenix

Alex-OPTIM, 2020-05-12:
How have you resolved the issue? I'm having the same problem.

wongyiuhang, 2020-09-02:
I've also encountered the same issue while building the NodeJS wrapper to librealsense 2.38.1 with Visual Studio. I successfully build the node wrapper simply with npm install

```
cd <librealsense_dir>\wrappers\nodejs
npm i --vs-configuration=Release
```

Reminder!
I have tried a few times and failed to build with NodeJS 14.x.x. At last, node_librealsense.node can be successfully built after I downgraded NodeJS to version 10.x.x (10.22.0, to be exact). Hope this can save a few hours of Googling from anyone in the future.