realsenseai/librealsense #4885: How to get timestamp when using multi camera in python?

shadowdk3, 2019-09-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.11.100 | 
| Operating System & Version |  Linux16.04  | 
| Kernel Version (Linux Only)    |  4.15.0-60                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.27.0                         | 
| Language                            |  python                        | 
| Segment			|  others                         | 

### Issue Description
I setup two D435 in order to sync two video together. I checked and verify two cameras are synchronize in SDK by using timestamp . But I would like to check timestamp in python.
How can I check timestamp in python with multi cameras?
`get_timestamp()` is not available in `frames_devices = device_manager.poll_frames()`
Thank you very much.

lramati, 2019-09-18:
Could you provide a little more information? I'm not sure what the device manager class or the poll_frames functions are.
In general the Python API is nearly 1-to-1 with the C++ API, so you should be able to do it in Python the same way you did it in C++.
If there's a gap in the Python bindings, please let me know and I'll try to get them wrapped in time for the next release.

shadowdk3, 2019-09-19:
@lramati Thanks for your reply.
I follow example https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/box_dimensioner_multicam_demo.py to get multi video.
Get all device video in `poll_frames` function, https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/realsense_device_manager.py

lramati, 2019-09-19:
if you look at the source of the function you linked, the object returned by `poll_frames()` is a dictionary of dictionaries of pyrealsense2 frames. you'd need to access those inner frames and call `get_timestamp()` on them

shadowdk3, 2019-09-26:
Finally, I success to use `wait_for_frames()` and get timestamp from frame. I close this Issues. Thanks for help.

closed 2019-09-26 by shadowdk3