realsenseai/librealsense #4887: Stereo Vision - absolute depth calculation?

arrowperfecto, 2019-09-18:
| Camera Model | D435 | 
| Segment | Stereo Depth | 

I am reviewing the different models of depth cameras.  I have a question about the "stereo vision".  Based on what I am reading in the detailed data [sheet](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Intel-RealSense-D40-Series-Datasheet.pdf) the depth calculation is not absolute depth, but a correlated value based on data from the two imagers.  Is this correct?  Can someone better explain how the depth is calculated?
Thanks,
Perfecto

agrunnet, 2019-09-18:
Correct. Please check out this explanation. We use triangulation. 
https://www.google.com/amp/s/www.intelrealsense.com/stereo-depth-vision-basics/amp/

arrowperfecto, 2019-09-18:
OK, thank you for the link.

closed 2019-09-25 by dorodnic