realsenseai/librealsense #4888: SR305 - imager and illumination

arrowperfecto, 2019-09-18:
Hi,
What imaging sensor is used in the SR305 (imager model)?  Is the illumination done by a VCSEL?
Thanks,
Perfecto

agrunnet, 2019-09-18:
It is a custom high speed imager. 
The laser is not VCSEL for this unit. 
The technology is coded light. Effectively it creates a line that scans very fast. The line is turned on and off in a specific pattern. An image is captured 10x faster than actually frame rate for depth. These 10 frames are exposed with different line patterns and depth is calculated based on all these subframes.

RealSenseCustomerSupport, 2019-10-08:
----------------------------------------------

Hi arrowperfecto,
Do you need further help on this topic?

arrowperfecto, 2019-10-08:
> CAUTION: This email originated from outside of the organization. This message might not be safe, use caution in opening it. If in doubt, do not open the attachment nor links in the message.

You can close this. Thank you.

closed 2019-10-09 by RealSenseCustomerSupport