realsenseai/librealsense #4890: Fails to turn on stereo module on MacOs

andreas-eg, 2019-09-18:
"Backend in rs2_open_multiple(sensor:0x7fd1025a9410, profiles:0x7fd10762b970, count:1):
fail to start streaming. Last Error: Operation timed out"

I got this on my mac after cmake the sdk's from source then using Xcode to build it.

The RGB camera is working.

I got the error when trying to enable the Stereo Module in realsense-viewer.

Console logged this message:
 18/09 22:47:02,682 ERROR [0x10a4295c0] (types.h:243) fail to start streaming. Last Error: Operation timed out
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   MacOS Mojave 10.14.6 | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | Mac OS   |
| SDK Version                            |  got latest srcs and built it }                          | 
| Language                            | realsense-viewer.                   | 
| Segment			|  others                           | 

### Issue Description
Debug op code: 15
Advanced mode: YES
Product ID = 0B3A
USB type descriptor 3.2

When clicking, Output log is saying "Starting streaming of Depth..." then the error message occurs.

kjacks21, 2019-10-07:
I'm also experiencing this issue on Mac OS, firmware version 05.11.15.00.

RealSenseCustomerSupport, 2020-02-10:
----------------------------------------------

@kjacks21 Seen from "Product ID = 0B3A", are you using D435i when get the issue?

RealSenseCustomerSupport, 2020-02-26:
----------------------------------------------

@kjacks21 Could you please update if you still need support on this? Thanks!

kjacks21, 2020-03-31:
@RealSenseCustomerSupport No, using D435. I do not currently need support for this. Thanks!

RealSenseCustomerSupport, 2020-03-31:
----------------------------------------------

@kjacks21 Thanks for your update! Will close this accordingly.

closed 2020-03-31 by RealSenseCustomerSupport