realsenseai/librealsense #4892: What does Tracking Module 1 (TM1) refer to?

antoan, 2019-09-19:
In the Prerequisites section of the install guide we have:
"TM1-specific:

Tracking Module requires hid_sensor_custom kernel module to operate properly. Due to TM1's power-up sequence constrains, this driver is required to be loaded during boot for the HW to be properly initialized." 

Not clear if this applies to me (T56 & D435i)

https://github.com/IntelRealSense/librealsense/blob/d6f6be84b46190c8c84c95f6ac279d239320fcda/doc/installation.md

dorodnic, 2019-09-19:
Hi @antoan
It does not. RealSense family is much broader than the handful of peripherals available for individual purchase. Tm1 refers to D400 with an IMU and a fish-eye lens, used by some of the bigger customers. 
Sorry if the instructions are a bit confusing, we keep since set of instructions and installation procedures for all RealSense customers.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

antoan, 2019-09-19:
Much appreciated, thanks.

closed 2019-09-19 by antoan