realsenseai/librealsense #4893: cols

nirshemesh, 2019-09-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

ev-mp, 2019-09-23:
@nirshemesh , please elaborate the issue.

nirshemesh, 2019-09-23:
Hi Intel,
My goal is to have **accurate** point cloud from up to meter of a small object (3*3*3[cm])
While working with the most updated FW and SDK (for D415). I found the following disturbing issue although the camera is calibrated according to the user guide:
1) Using point cloud example in Matlab - the point cloud is very noisy and not accurate experiencing "breathing" effect over time.
2) The depth is also noisy - so I used temporal averaging per depth per pixel and I achieved "stability" but while doing so on the point cloud map - I found it still noisy.
3) can you suggest on a filter solution + example from point_cloud.m file how to apply it so it will be accurate ? Or other solution to overcome this need ?

Thanks

Nir

dorodnic, 2019-09-24:
https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance

nirshemesh, 2019-09-29:
Done that already,

Looks like the dynamic calibration tools doesnt progress (% value stays on 0).
I tried both Letter page and Mobile phone as a target pattern - with no success (I checked the dimensions as requested).

I follow the instruction on the link above and still - breathing pointcloud achieved (Matlab example).

I"m using D415 with the most updated SW & FW.

Last - I dont have a specific Matlab example for pointcloud that contain temporal filter in it - maybe that should help ?

Distance to the wall (80[cm]) and the target around 60[cm].

Will be glade to have your solution

Thanks
-Nir

RealSenseCustomerSupport, 2019-10-31:
----------------------------------------------

@nirshemesh Sorry for late response. Would like to know if you still have issue. For dynamic calibration, what OS did you try the tool? Ubuntu or Windows?

nirshemesh, 2019-10-31:
Yes, I still have the issue. I"m using Windows 10. in addition all the calibration guides I"ve used doesnt solve the problem of "breathing" point cloud - its not stable at all. your help is appreciated.

Nir

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

@Nir Could you please provide some snapshots of the point cloud you see the issue?  And what's your SDK and FW version now? Thanks!

RealSenseCustomerSupport, 2019-11-07:
----------------------------------------------

@Nir Could you provide more details about the issue so that we can continue to look into it? Thanks!

nirshemesh, 2019-11-07:
The FW version is 5.11.11.100 (D415) SDK version is 2.0. (Windows 10).

The average temporal error of the point cloud of a flat wall located around 80 [cm] from the camera is around 8[mm]. I want to understand if this is the accuracy I should expect from this model ? 

My need to have accuracy of around 1mm...

As i said the pointcloud is "breathing" overtime (temporal),

Thanks

Nir

RealSenseCustomerSupport, 2019-11-11:
----------------------------------------------

@nirshemesh According to section 4.7 in the datasheet https://www.intelrealsense.com/wp-content/uploads/2019/10/Intel-RealSense-D400-Series-Datasheet-Oct-2019.pdf , the temporal noise is ≤ 1% for D400 cameras(up to 2 Meters and 80% FOV). You can refer to https://dev.intelrealsense.com/docs/depth-post-processing and try post-processing to see if it helps. Thanks!

RealSenseCustomerSupport, 2019-12-23:
----------------------------------------------

@nirshemesh Sorry that we didn't get your feedback for several weeks. Will close the ticket at this point. Please create another one if you still have issues or questions. Thanks!

closed 2019-12-23 by RealSenseCustomerSupport