realsenseai/librealsense #4894: the accuracy of the depth about the d400

hpf9017, 2019-09-20:
I'm trying to get the best performance of depth about d435i according to
[https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf](url)

in 3. Verify performance regularly on a flat wall or target:
e. Understand theoretical limit:

[image]

[image]

from the picture,
is it that mean when measured from the 1 meter away with d435i
the depth I can get is 1 meter ± 3mm?

MartyG-RealSense, 2019-09-20:
Intel members JesusG and agrunnet provide information about the mm accuracy of the 400 Series cameras in the link below.

https://github.com/IntelRealSense/librealsense/issues/999

RealSenseCustomerSupport, 2019-10-09:
----------------------------------------------

Closing this if there is no further question.
Thank you.

closed 2019-10-18 by RealSenseCustomerSupport