realsenseai/librealsense #4895: [ERROR]: (...) API version mismatch: librealsense.so was compiled with API version 2.28.0 but the application was compiled with 2.25.1! Make sure correct version of the library is installed (make install)

Daniel2401, 2019-09-20:
|Required Info |  |
|-- | --
|Camera Model | D435|
|Firmware Version | 05.11.06.250|
|Operating System & Version | Ubuntu 16.04.6 LTS|
|Kernel Version (Linux Only) | 4.15.0-64-generic|
|Platform | PC|
|SDK Version |2|
| Language                            |  python                          | 
| Segment			|  Robot                          | 

### Issue Description
I have a similar problem to #1724, but with different SDK versions.
`
[ERROR]: (...) API version mismatch: librealsense.so was compiled with API version 2.28.0 but the application was compiled with 2.25.1! Make sure correct version of the library is installed (make install)`

I followed https://github.com/IntelRealSense/librealsense/tree/development/wrappers/python but still the above error still occurs.

My steps were:

1. `mkdir reposerver`
2. `cd reposerver`
3. `git clone https://github.com/IntelRealSense/librealsense.git`
4. `cd librealsense`
5. `mkdir build`
6. `cd build`
7. `cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python`
8. `make -j4`
9. `sudo make install`
10. `export PYTHONPATH=$PYTHONPATH:/usr/local/lib`

lramati, 2019-09-22:
It seems like python is picking up an old build of pyrealsense2 which was compiled with an older version of librealsense.
You should find where this module is and manually remove it, along with any old versions of librealsense and then recompile when they can no longer interfere. running `sudo make uninstall` to remove the system-wide librealsense install and then trying to import pyrealsense should generate some errors which can start you off in the right direction

RealSenseCustomerSupport, 2019-10-10:
----------------------------------------------

Hi @Daniel2401,

Do you need any further clarifications regarding this topic?

Daniel2401, 2019-10-11:
Hi,

thanks for your help @lramati.
I haven't gotten to it since, but will let you know whether I need more help or not once I had the chance to @RealSenseCustomerSupport.

RealSenseCustomerSupport, 2019-10-17:
----------------------------------------------

Hi @Daniel2401,

Did you have a chance to apply the solution @ lramati has proposed?

Thank you!

Daniel2401, 2019-10-24:
Hi,

I found these three files outside the directory mentioned in the original post when searching for "pyrealsense2" in usr/local/lib. So this should be the old build to remove right?
pyrealsense2.so
pyrealsense2.so.2.28
pyrealsense2.so.2.28.0

In which directory do I have to `sudo make uninstall librealsense2`, because right now I get `make: *** No rule to make target 'uninstall'.  Stop.`

RealSenseCustomerSupport, 2019-10-25:
----------------------------------------------

Hi @Daniel2401,

Yes, you can remove these files and run sudo make uninstall from build folder of librealsense root directory.

Thank you!

Daniel2401, 2019-10-28:
Hi,

got it working again. Thanks everyone!

closed 2019-10-28 by Daniel2401