realsenseai/librealsense #4897: Fatal unmanaged exception after multiple rs2::device affectations

blackccpie, 2019-09-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 5.11.1 | 
| Operating System & Version |   MacOS  | 
| Platform                                 | laptop |
| SDK Version                            |  2.28.0                          | 
| Language                            |  C++                          | 
| Segment			|  others                        | 

### Issue Description

When I repeat multiple consecutive playback start/stop in a loop, in my case I always encounter unmanaged fatal exception after a few iterations. The code I'm using is as simple as that:

```cpp
rs2::device device;
rs2::pipeline pipe;

for(int i=0; i<500; i++)
{
	std::cout << "#" << i << std::endl;
	rs2::config config;
	rs2::pipeline_profile pipeline_profile;
	config.enable_device_from_file( "my_bag.bag", true );
	pipeline_profile = pipe.start( config );
	device = pipeline_profile.get_device(); // crashes in rs2::device assignation
	pipe.stop();
}
```

Notice that if I comment the device assignation line the problem cannot be reproduced.

Could you confirm and help me resolve the problem?

Regards

Albert

MartyG-RealSense, 2020-03-04:
I apologise that you did not receive a response to this case until now.  

Do you still require assistance with it please or is it okay to close the case?   Thanks!

jillemash, 2020-03-04:
Hi,

On behalf of @blackccpie , I've reproduced the issue with the following version:

- **2.29.0**
- **2.32.1**

Program exits with status 11, for random values of i, but the stack seems to be always the same:

```
_pthread_cond_wait 0x00007fff6c85722a
<unknown> 0x000080036976fcc1
dispatcher::stop() 0x0000000106773258
librealsense::playback_device::~playback_device() 0x00000001066534dc
std::__1::shared_ptr<librealsense::auto_exposure_state>::~shared_ptr() 0x000000010655d2c3
librealsense::readonly_device_info::~readonly_device_info() 0x000000010683d0e6
std::__1::shared_ptr<librealsense::auto_exposure_state>::~shared_ptr() 0x000000010655d2c3
rs2_delete_device 0x00000001068034e9
std::__1::__shared_ptr_pointer<rs2_device*, void (*)(rs2_device*), std::__1::allocator<rs2_device> >::__on_zero_shared() memory:3640
std::__1::__shared_count::__release_shared() memory:3544
std::__1::__shared_weak_count::__release_shared() memory:3586
std::__1::shared_ptr<rs2_context>::~shared_ptr() memory:4522
std::__1::shared_ptr<rs2_context>::~shared_ptr() memory:4520
std::__1::shared_ptr<rs2_device>::reset() memory:4657
rs2::device::operator=(std::__1::shared_ptr<rs2_device>) rs_device.hpp:100
rs2::device::operator=(rs2::device const&) rs_device.hpp:106
main main.cpp:15
start 0x00007fff6c64d7fd 
```
(main.cpp:15 is the assign statement mentioned in the first message)

So this case should not be closed.

Thanks.

JM

MacOS Catalina 10.15.3

blackccpie, 2020-03-05:
@MartyG-RealSense : thanks @jillemash, so the bug is still present in the last release. 
I made a quick test on my Windows10 system, and I could not reproduce the issue (but as we're planning mac releases, solving this bug is quite important for us).

Regards

MartyG-RealSense, 2020-03-05:
There is already a Known Issue for starting after a stop on Mac for the T265 in the SDK release notes.  Perhaps there is a related issue with the 400 Series cameras.

**[T265][Mac] - Start after stop is not working on Mac with the T265 camera**

This suggests that the issue is on the developer team's bug-fix radar, at least in terms of the T265.  I will tag @dorodnic in on this discussion and also add the 'bug' label to it to aid tracking by the RealSense team.

MartyG-RealSense, 2020-06-05:
Case closed due to lack of recent cases of the error during start-stop, and bug-fixes for IR stream start-stop in firmware **5.12.5.**

blackccpie, 2020-06-09:
@MartyG-RealSense : I'm a little surprised to see this issue closed. It happens the bug is still there on MacOS with the latest realsense release available on brew (2.34). Also this bug has nothing to do with the hardware firmware, as it can be reproduced using a replay file, without any plugged camera. It's really easy to reproduce given the code provided in the issue description, so I guess it should be easier to investigate than the other hard to reproduce issues of the tracker.

MartyG-RealSense, 2020-06-09:
I do not have a Mac computer available or a BREW setup to exactly test and try to replicate your scenario, unfortunately.  There can also be a range of causes for errors that result from repeated stop-starts (for example, the T265 camera's stop-start error is still on the SDK's 'Known Issues' list of unresolved problems in the SDK release notes. 

 I would recommend waiting for 2.35.0 to become available for your setup and see if that corrects the problem.  I apologise that I do not have more than that to offer right now.

ev-mp, 2021-08-17:
@blackccpie , there were multiple refactoring implemented in the `playback_device` class in the recent releases.
We believe that the reported issue is no longer relevevant

blackccpie, 2021-08-30:
@ev-mp : after updating librealsense to the latest available release on my mac's brew, `v2.48.0`, the behavior has changed, as the crash is not systematic anymore, but unfortunately it sometimes still occurs. I'll give it a try with `v2.49.0` when available.

closed 2023-07-18 by MartyG-RealSense