realsenseai/librealsense #4898: What would cause an unknown error exception to be thrown when calling get_option(RS2_OPTION_GAIN) in Android?

davidsamson, 2019-09-20:
I have upgraded my RS415 and RS435 to the latest firmware and I am using librealsense2.28.0. Both of these cameras are throwing an exception when attempting to get/set gain in Android. Get/set exposure option works just fine. I have caught and printed out the exception error, but it is no help. Also, I drilled down in the RS SDK source code and I can see that this error is caught by a catch(...) in translate_exception. I am running my Android app on a Galaxy S10e and an HTC U12+. Also, I am running the exact same code (compiled for Windows) with the cameras connected to a Windows 10 machine with no problems.

Gilaadb, 2019-09-24:
Hi @davidsamson, I've found your issue report while in the process of opening a similar issue for a different camera model. If you are interested in a workaround, please try to add a sleep of 100 ms before setting gain. Of course, if it helps, you can optimize this solution by setting the gain optimistically and only if failing then to wait a short while and retry.
I did not respond about the "get_option" variant because I stumbled upon this issue only when setting the gain.

RealSenseCustomerSupport, 2019-10-25:
----------------------------------------------

@davidsamson Could you please tell which FW used when you get the issue?

RealSenseCustomerSupport, 2020-02-14:
----------------------------------------------

@davidsamson I tried with the latest SDK version with FW 5.12.1 and couldn't reproduce. Could you please try the latest SDK and FW version to see if you still have the issue? Looking forward to your update. Thanks!

davidsamson, 2020-02-17:
Thanks, I have this fixed now with the latest firmware.

RealSenseCustomerSupport, 2020-02-18:
----------------------------------------------

@davidsamson Glad to hear that fixed. Thanks for the feedback!

closed 2020-02-18 by RealSenseCustomerSupport