realsenseai/librealsense #4899: Raspberry PI 4 Buster and D435i - Error when trie use motion data

rofreli, 2019-09-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435I                                                      | 
| Firmware Version                   | 05.11.11.100                                            | 
| Operating System & Version |   Raspbian Buster                                    | 
| Kernel Version (Linux Only)    |  4.19.73-v7l+                                          | 
| Platform                                 | Raspberry Pi                                           |
| SDK Version                            | 2.28.0                                                     | 

### Issue Description
When I run realsense-viewer I got Error type: No data to return (-21) and when I try to execute Motion Module realsense-viewer close with BUS Error. Others modules work fine, just Motion Module with Accelerometer and Gyro broke.

MartyG-RealSense, 2019-09-21:
Another RealSense user on the Intel Support forum recently also had problems with Pi 4 and Raspian Buster when activating the IMU but otherwise had no problems with the camera's other modes.  They also experienced the 'no data to return (-21)' error.

https://forums.intel.com/s/question/0D50P00004RoUHqSAN/d435i-motion-module-force-pause-when-turning-on-imu?language=en_US

I suggested trying installing Librealsense on their Pi using **libuvc-backend**, an installation method that is useful when the usual installation instructions do not work.  I have not heard back yet whether it worked for them though.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

rofreli, 2019-09-21:
I tried this step and I had the same result. After run this script, rs-enumerate-devices showed for me

Stream Profiles supported by Motion Module
 Supported modes:
    stream       resolution      fps       format   
    Gyro	 N/A		@ 200Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 400Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 63Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 250Hz	   MOTION_XYZ32F

But when I execute Motion Data in realsense-viewer or in my python code I have the same error

 21/09 19:26:19,733 WARNING [2918564928] (types.cpp:48) hwmon command 0x4f failed. Error type: No data to return (-21).

MartyG-RealSense, 2019-09-22:
Have you calibrated the D435i's IMU component please?  The D435i IMU, unlike the other sensing components, is not calibrated in the factory and has to be calibrated by the user after purchase.

https://www.intel.com/content/www/us/en/support/articles/000032303/emerging-technologies/intel-realsense-technology.html

rofreli, 2019-09-22:
Yes, I did the calibration in Windows, but when I do the same process in raspi I got the error 

Start interactive mode:
FOUND GYRO with fps=400
FOUND ACCEL with fps=250
-------------------------
*** Press ESC to Quit ***
-------------------------

Align to direction:  [ 0. -1.  0.]   Upright facing out
Erro no barramento

MartyG-RealSense, 2019-09-22:
'Erro no barramento' translates to "bus error".

I was not able to locate a solution to this particular problem - one of the Intel team members on this forum may be able to offer helpful advice.  If you have successfully completed the IMU calibration in Windows though then I would think that it would not be necessary to repeat it in Raspbian.

rofreli, 2019-09-22:
I will try to start the installation from a new image. If I have better luck, I write here. 
Thank you for your help!

rofreli, 2019-09-23:
I start from a new Image, did the steps:

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

With change in CMake [#4565](https://github.com/IntelRealSense/librealsense/issues/4565#issuecomment-518359584)

set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -latomic")

And execute with root privileges(sudo su) because I got an error "Need to be root" when the script run sudo udevadm control --reload-rules && udevadm trigger 

And the results are the same, I sill with problems in Motion data, someone has an idea or a raspbian buster image without this problem?

frank26080115, 2019-09-24:
Hey I've gotten the same problems, D435i, Raspbian Buster, master branch of librealsense. Many messages saying that the bulk_transfer is failing, I tried many things, including recompiling the code so that it does a call to libusb_claim_interface, but that fails too, I even tried to libusb_detach_kernel_driver before libusb_claim_interface and even that is failing.

Patching the Rasbian Buster kernel didn't even help but it removed many dmesg UVC errors complaining about missing video formats, and it allows cheese to use the camera

rs-motion doesn't work, rtabmap doesn't work, ROS will give me errors unless I disable the IMU in the launch files.

Maybe, another process claimed those endpoints?

Marty has been helping me but too on the same problem but there was never a solution. I actually purchased a D415 instead just to ignore this problem.

By the way, I am using `sudo udevadm control --reload-rules && sudo udevadm trigger` and it gives no errors (keyword being the second sudo)

Hey @rofreli wanna chat for a bit outside of this github issue? We might be working on something similar. I've automated the entire installation of librealsense, ROS, and rtabmap on top of Raspbian Buster.

frank26080115, 2019-09-25:
Hey good news, I somehow got a built that works. I actually don't know what I did to fix it though... The last thing I did was try and fix a segfault that was unrelated.

I got some things to contribute, such as a patch file, and my CMakeCache.txt , if you need more, just ask, but the entire build directory is a whopping 1.7GB (includes all object files and such)

[patch_librealsense_ff2a291.txt](https://github.com/IntelRealSense/librealsense/files/3650368/patch_librealsense_ff2a291.txt)

[CMakeCache.txt](https://github.com/IntelRealSense/librealsense/files/3650371/CMakeCache.txt)

The patch file does include the automatic libusb_claim_interface, but previously it didn't work... I'm not sure why it worked today. It only patches FF2A291 though, and master is ahead now.

rofreli, 2019-09-25:
Did you made some changes in files like messenger-libusb.h and cpp, sensor.cpp ? I will try with this changes in my raspi and I tell you after.

frank26080115, 2019-09-25:
Yes, simply check out the hash ff2a29197056f240263c787f059ef4e561e6608c and then use `git apply patch_librealsense_ff2a291.txt` and it will edit all those files for you

The master branch changes too often for me to update the patch file for each commit in this repo. The previous release of realsense-viewer doesn't actually work on Raspbian Buster. So that's why it has to be ff2a291 or later.

rofreli, 2019-09-27:
Hi Frank , I tried your solution but I got the same error. I ..... I believe is something in raspbian kernel.

frank26080115, 2019-09-27:
Oh I've been doing builds with GCC 8 instead of GCC 6, I know the instructions said 6 but I am using 8, maybe that can help? (it's unlikely that this is the problem)

rofreli, 2019-10-11:
Does anyone have some news about this?

MartyG-RealSense, 2019-10-11:
Here's the latest on the problems with Pi 4 and the IMU:

https://github.com/IntelRealSense/librealsense/issues/4979#issuecomment-540456439

rofreli, 2019-10-14:
This [https://github.com/IntelRealSense/librealsense/pull/5043/commits](5043) works for me! I already finish simple tests, but it works!

RealSenseCustomerSupport, 2019-10-31:
----------------------------------------------

@rofreli Glad to hear that the issue resolved. Will close it accordingly.

closed 2019-10-31 by RealSenseCustomerSupport