realsenseai/librealsense #4901: Error trying to record .bag file with realsense viewer

sodom89, 2019-09-21:
[image]
| | |
|---|---|
|**librealsense**|2.28.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435|
|**Serial Number**|831612070780|
|**Firmware Version**|05.11.11.100|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|830513021085|

If I am trying to record the data streams of my D435 or D435i, I get the RealSense error calling like provided in the image every time. If I'm using Windows with the same settings, this error doesn't occur. Any suggestions how to solve this problem?

MartyG-RealSense, 2019-09-21:
Could you try setting a different path to save the file to please to make sure that the path you are using does not have permissions that prevent the writing of a file to it?  Thanks!

https://github.com/IntelRealSense/librealsense/issues/3680

sodom89, 2019-09-27:
If tested your suggestion and it works. Thanks

closed 2019-09-27 by sodom89

erv98, 2020-03-09:
> Could you try setting a different path to save the file to please to make sure that the path you are using does not have permissions that prevent the writing of a file to it? Thanks!
> 
> #3680

how do you do this? Im getting the same error as the one above. I want to save and stream record the two IR footage

MartyG-RealSense, 2020-03-09:
First, please go to the gear-wheel icon in the top corner of the Viewer, left-click on it and select 'Settings' from its menu.

[image]

On the options window, please select the 'Playback and Record' options tab if it is not selected already.  It will then allow you to enter a custom save path for recordings into   When you have entered the custom path into the text box, please click the 'Apply' button beneath it.

[image]