realsenseai/librealsense #4903: librealsense Android 2.23.0: occasionally  received signal SIGSEGV, segmentation fault.

wbing-xu, 2019-09-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | newest | 
| Operating System & Version |   Android8.1  | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | RK3399 |
| SDK Version                            | 2.23}                          | 
| Language                            | c++   | 
| Segment			|  libuvc                          | 

### Issue Description

When I open device and get stream, occasionally received signal SIGSEGV, Segmentation fault.

The final error log:
[image]

My code looks like this:
```cpp
void RGBDRealSense::openDevice(rs2::device &dev)
{
    RealSenseInfo real_sense_info;
    dev.hardware_reset();

    rs2::config cfg;
    auto serial = dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER);
    auto physical_pport = dev.get_info(RS2_CAMERA_INFO_PHYSICAL_PORT);
    cfg.enable_stream(RS2_STREAM_DEPTH, width_, height_, RS2_FORMAT_Z16, fps_);
    cfg.enable_stream(RS2_STREAM_INFRARED, 1, width_, height_, RS2_FORMAT_Y8, fps_);

    LOGD("RGBDRealSense") << "starting rgbd:" << serial;
    cfg.enable_device(serial);
    
    auto pipe = std::make_unique<rs2::pipeline>(ctx);

    LOGD("RGBDRealSense") << "pipe starting";
    auto profile = pipe->start(cfg);
    LOGD("RGBDRealSense") << "profile get_stream";
    int id = profile.get_stream(RS2_STREAM_DEPTH).unique_id();
    LOGI("RGBDRealSense") << " rgbd open done " << serial << " with  " << id;
    serial_id.insert(pair<int, std::string>(id, serial));

    rs2::device selected_device = profile.get_device();
    auto depth_sensor = selected_device.first<rs2::depth_sensor>();
    depth_sensor.set_option(rs2_option::RS2_OPTION_VISUAL_PRESET, rs2_rs400_visual_preset::RS2_RS400_VISUAL_PRESET_HIGH_ACCURACY);
().get_option(rs2_option::RS2_OPTION_VISUAL_PRESET);

    auto depth_scale = depth_sensor.get_depth_scale();
    LOGI("RGBDRealSense") << " rgbd set option depth_sensor depth_scale" << depth_scale;
   //......
   //No memory operations performed
   //.....
    LOGD("RGBDRealSense") << "end";
}
```
my progress log is
[image]

This is the complete rgbd log

[_rgbd_sensor_log.zip](https://github.com/IntelRealSense/librealsense/files/3638635/_rgbd_sensor_log.zip)

RealSenseCustomerSupport, 2019-10-25:
----------------------------------------------

@wbing-xu Did you get chance to try the latest SDK and FW to see if you still have issue? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-11-11:
----------------------------------------------

@@wbing-xu Could you please update? Thanks!

RealSenseCustomerSupport, 2019-11-22:
----------------------------------------------

@wbing-x Sorry that we didn't get response from you for weeks. Will close this issue at this point. Please create new one if you still have questions/issues. Thanks!

closed 2019-11-22 by RealSenseCustomerSupport