realsenseai/librealsense #4904: Is there a recommended hub for commercial use?

tylandercasper, 2019-09-21:
We are building a tracking system that uses 2 of your cameras which is intended for consumer use. 
We're making a pretty case for it and would like to put a hub inside so that the user does not have to run 2 usb cables. 
Unfortunately, almost all of the hubs we have tried have proven unreliable and caused a great deal of issues. 

If cost is not an issue, but knowing it will still work in 2 years is:  Is there a hub or hub circuit board that you can recommend we use?
Are there any specs or other features that would help us find one?

Thanks,
Ty

MartyG-RealSense, 2019-09-21:
In Intel's white paper document about using multiple RealSense cameras, they tested with an AmazonBasics powered hub and provided the product link below in the paper.

https://www.amazon.com/dp/B00DQFGH80/

The full paper is available here if you have not seen it:

https://www.intel.co.uk/content/www/uk/en/support/articles/000028140/emerging-technologies/intel-realsense-technology.html

I have seen reports of RealSense users having problems with USB 3.2 hubs, whereas USB 3.1 hubs work for them.

I wonder if a tidy way to design your product would be to base it around an Intel NUC mini PC kit.  They are very small, you can choose your own components and they have internal USB ports.

https://forums.intel.com/s/question/0D50P0000490VcOSAU/new-budgetprice-intel-nuc-models?language=en_US

RealSenseCustomerSupport, 2019-10-02:
----------------------------------------------

Can't agree more on Marty's comment.

Can I ask how you run the two cameras via the USB hub?
- What is the image size, frame rate, and how many videos streaming running at same time? e.g. depth, IR, and RGB are running at same time?
- Were the two cameras running on Multi-cam Sync mode? or independently (run freely) ?
- What about the firmware version? (by the way, there is new firmware just released FW5.11.15.100, would you mind to give a shot? (download it from here: https://www.intelrealsense.com/developers/)
- How does your host platform look like?

Thanks

**Please be informed that we will close this ticket in seven (7) days if we have not heard back from you.

closed 2019-10-17 by RealSenseCustomerSupport