realsenseai/librealsense #4908: No +realsense folder created when compiling Matlab wrappers

PascalYIN, 2019-09-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 05.11.11.100 | 
| Operating System & Version |   {Win (10)}  | 
| Platform                                 | PC |
| SDK Version                            |  { 2.28.0 }                          | 
| Language                            |  {C++/Matlab}                          | 
| Segment			|  {others }                          | 

Hello,

I built Realsense SDK from the source code since for unknown reason the download of Windows installer always failed on my desktop. Following the installation guide of the Matlab wrapper I tried to build the librealsense_mex target with Visual Studion 2017. The compilation finished without error, but I was not able to find the +realsense folder anywhere in the build folder. I'm not sure if I did something wrong but I have literaly followed the Matlab wrapper and looked up the folder everywhere and it just wasn't there. In the build/Release folder there is only a "librealsense_mex.mexw64". Could you please tell me where I did wrong. I was using Matlab 2018b instead of 2017b, not sure if that changes anything.

Thanks!

lramati, 2019-09-22:
No. It's simply that the documentation is a bit ambiguous and nobody caught it before now.
You'll notice that in the matlab folder we provide `librealsense_mex.vcxproj`, which was set up to automatically copy everything necessary into a folder called `+realsense`. This project is not up to date and isn't the recommended build method anymore, however, the project generated by CMake doesn't do this step yet.
All you need to do is copy all the `*.m` files under `wrappers/matlab` and also `librealsense_mex.mexw64` from `build/Release` into a folder named `+realsense` and then continue with the installation guide. Please note that if you compiled librealsense as a shared library, you'll need to copy `realsense2.dll` into the `+realsense` folder as well

PascalYIN, 2019-09-24:
Thanks for your response.

closed 2019-09-24 by PascalYIN