realsenseai/librealsense #4911: Add instructions to resolve Windows driver error from #2612.

CaleMcCollough, 2019-09-22:
# Windows Bug Fix

## Problem

There is a USB Cam Overflow error with my Realsense D435i and the issue ticket that regards this issue has bad instructions that I discovered the solution for. If you install the Realsense SDK, I just tried 2.28.1.1106 on an Intel laptop, then you plug in the Realsense, Windows will use a very old driver fro 2006, and you need to upgrade the driver, and the directions are missing as well as the link to the UWP Drivers that can be found in Intel RealSense D400 Series UWP Driver v6.1.160.14.zip. My D435i will not work on my AMD mobo without using a USB3 hub with its own power supply because it will cause my mobo's USB3 chipset to a malfunction that requires a reboot to fix, and the camera freezes with the USB Cam Overflow error. This does not happen on my Intel laptop. I either have a defective unit or it is drawing too much current, which might be why it only works with USB C cables that can't supply the current.

## Solution

### Hardware

All of these cable and USB Cam overflow bugs are under-current power issues. Non-Intel Users should be directed to purchase an external USB hub with USB charging port that can handle at least X amp per channel where X is detereed by Intel, but the USB 3 spec is 900mA by my recollection, but my AMD mobo should supply that much current, so we should probably say 1.5A minimum recommend power.

### Link to Drivers on Download Page

Users need a link to the Windows Directions on the download page with additional links for drivers for other Realsense models.

### Windows Directions 

1. Download the [RealSense D400 series serial UWP driver](https://downloadcenter.intel.com/download/27799/Intel-RealSense-D400-Series-Universal-Windows-Platform-UWP-Driver-for-Windows-10) and extract it to folder Foo.

2. Open the device manager then right-click on Cameras->Intel(R) RealSense(TM) Depth Camera 435i RGB->Driver->Update Driver.

3.) Click on Update Driver then Search my Computer for Driver Software then point it to Foo, and click Next.

4.) Repeat steps 1-3 for the depth sensor camera.

RealSenseCustomerSupport, 2019-10-21:
----------------------------------------------

Hi @CaleMcCollough,

Please provide more details about "an old driver from 2006" you have mentioned. Along with that information,please describe the exact scenario that leads to the problem.

Thank you!

RealSenseCustomerSupport, 2020-02-24:
----------------------------------------------

@CaleMcCollough,

Are you still interested to continue discussion about this topic?

closed 2020-03-02 by RealSenseCustomerSupport