realsenseai/librealsense #4912: Error trying to patch uvcvideo part of installation tutorial

grantsadie88, 2019-09-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 } | 
| Operating System & Version |   {Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  (4.15.0-64-generic)                                         | 
| SDK Version                            |  { legacy }                          | 

### Issue Description

Hello, I'm trying to install librealsense for my R200 camera and when i get to the step that asks me to run the line  ```./scripts/patch-uvcvideo-16.04.simple.sh``` I receive an error that states ```./scripts/patch-uvcvideo-16.04.simple.sh: line 15: ../scripts/realsense-camera-formats_ubuntu16.patch: No such file or directory```
How can i fix this?

ev-mp, 2019-09-23:
@grantsadie88 hello,
The script that you mention was deprecated and it is not part of the [legacy documentation](https://github.com/IntelRealSense/librealsense/blob/legacy/doc/installation.md).

The legacy branch maintenance handles mainly code compatibility issues.To get support for the the most recent Ubuntu kernels you need to reuse Librealsense2 patches as those are backward-compatible with all the legacy devices.

Download the code and patches for Librealsense2 (found in `master/development` branches) and run the embedded script from the [manual](https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md) to patch the kernel:
>./scripts/patch-realsense-ubuntu-lts.sh

Then you can continue using the legacy branch with R200.

grantsadie88, 2019-09-23:
I understand, thanks for your help, I'm quite new to this. I've done as you've said and followed the rest of the installation procedures, however I'm still receiving this error when i try open realsense-viewer with my R200 attached.

``` 23/09 18:29:11,898 WARNING [140553705617152] (types.cpp:48) set_pu(id=3) failed! Last Error: Input/output error
 23/09 18:29:11,899 WARNING [140553705617152] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Controls exposure time of color camera. Setting any value will disable auto exposure
 23/09 18:29:16,984 WARNING [140553705617152] (types.cpp:48) set_pu(id=9) failed! Last Error: Input/output error
 23/09 18:29:16,984 WARNING [140553705617152] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance
```

What might this mean?

ev-mp, 2019-09-23:
@grantsadie88 , the `realsense-viewer` is part of Librealsense2 and is not compatible with R200.
The viewer tool for Librealsense**1** is named `cpp-config-ui`. In practice all Librealsense1 tools/examples are named `c-..` or `cpp-...`.

I repost the build instruction for legacy branch for convenience - https://github.com/IntelRealSense/librealsense/blob/legacy/doc/installation.md

grantsadie88, 2019-09-23:
Ah got it. Thank you so much for your help.

closed 2019-09-23 by grantsadie88