realsenseai/librealsense #4913: Depth Quality Tool couldn't "Start_record."

KC176, 2019-09-23:
| | |
|---|---|
|**librealsense**|2.28.1 RELEASE|
|**OS**|Windows|

I meet a problem about saving images continuously. I want to save 100 images continuously using D435i and D415. When I click "Start_record", Depth Quality Tool will be interrupted. How to resolve it?

Firmware :　05.11.11.100
Depth Quality Tool : v2.28.1

[image]

ev-mp, 2019-09-23:
@KC176 , thank you for the report.
I can confirm it is a bug with the SDK, will be handled in next releases.

KC176, 2019-09-24:
@ev-mp , 
Ok, thank you very much!

RealSenseCustomerSupport, 2019-11-11:
----------------------------------------------

Hi @KC176,

Please use Depth Quality Tool v2.30.0 and FW v5.11.15.0 for your task and let us know if you need additional help with this issue.

Thank you!

PR #4967 (2019-10-23): DQT/Viewer metrics record fixes and improvements
Fixes #4913, #4948
closed 2019-11-19 by dorodnic