realsenseai/librealsense #4916: RealSense Viewer does not detect camera on Linux anymore - as it did before

neilyoung, 2019-09-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | no info | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.15.0-64-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            | 4.15.0-64-generic                        | 
| Language                            |                           | 
| Segment			|                        | 

### Issue Description
While having RealSense Viewer running on Windows 10 on the same machine (Asus Vivobook S) and while having had it also running on Ubuntu 16.04 RealSense Viewer is no longer able to detect the camera. `lsusb` is detecting the Intel device when plugged, but the app does not.

Is there any means to approach the problem?

ev-mp, 2019-09-23:
@neilyoung hello, 
What is the SDK version that you use ?
Can you provide the kernel/application logs when connecting the camera to PC and invoking `rs-enumerate-devices` ? (`dmesg -wT`) ?

neilyoung, 2019-09-23:
@ev-mp Thanks for the follow up. I'm currently doing again a time-intensive thing  on my Windows partition, but I'm able to provide the logs later. In the meantime it might be of interest, that I'm constantly getting this error on installation (tried several times):

```
Building only for 4.15.0-64-generic
Building initial module for 4.15.0-64-generic
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/librealsense2-dkms.0.crash'
Error! Bad return status for module build on kernel: 4.15.0-64-generic (x86_64)
Consult /var/lib/dkms/librealsense2-dkms/1.3.5/build/make.log for more information.
```

and the mentioned make.log reports things like this:

```
decades@ubuntu:~$ cat /var/lib/dkms/librealsense2-dkms/1.3.5/build/make.log 
DKMS make.log for librealsense2-dkms-1.3.5 for kernel 4.15.0-64-generic (x86_64)
Sat Sep 21 10:38:16 CEST 2019
make: Entering directory '/usr/src/linux-headers-4.15.0-64-generic'
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_driver.o
In file included from ./arch/x86/include/asm/atomic.h:5:0,
                 from ./include/linux/atomic.h:5,
                 from /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_driver.c:14:
./arch/x86/include/asm/qspinlock.h: In function ‘native_queued_spin_unlock’:
./arch/x86/include/asm/qspinlock.h:42:25: error: ‘struct qspinlock’ has no member named ‘locked’
  smp_store_release(&lock->locked, 0);
                         ^
./include/linux/compiler.h:304:9: note: in definition of macro ‘__compiletime_assert’
   if (!(condition))     \
         ^
./include/linux/compiler.h:324:2: note: in expansion of macro ‘_compiletime_assert’
  _compiletime_assert(condition, msg, __compiletime_assert_, __LINE__)
  ^
./include/linux/compiler.h:327:2: note: in expansion of macro ‘compiletime_assert’
  compiletime_assert(__native_word(t),    \
  ^
./include/linux/compiler.h:327:21: note: in expansion of macro ‘__native_word’
  compiletime_assert(__native_word(t),    \
                     ^
./arch/x86/include/asm/barrier.h:97:2: note: in expansion of macro ‘compiletime_assert_atomic_type’
  compiletime_assert_atomic_type(*p);    \
  ^
/var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/../../../../include-overrides/asm-generic/barrier.h:157:33: note: in expansion of macro ‘__smp_store_release’
 #define smp_store_release(p, v) __smp_store_release(p, v)
                                 ^
./arch/x86/include/asm/qspinlock.h:42:2: note: in expansion of macro ‘smp_store_release’
  smp_store_release(&lock->locked, 0);
  ^
./arch/x86/include/asm/qspinlock.h:42:25: error: ‘struct qspinlock’ has no member named ‘locked’
  smp_store_release(&lock->locked, 0);
                         ^
./include/linux/compiler.h:304:9: note: in definition of macro ‘__compiletime_assert’
   if (!(condition))     \
         ^
./include/linux/compiler.h:324:2: note: in expansion of macro ‘_compiletime_assert’
  _compiletime_assert(condition, msg, __compiletime_assert_, __LINE__)
  ^
./include/linux/compiler.h:327:2: note: in expansion of macro ‘compiletime_assert’
  compiletime_assert(__native_word(t),    \
  ^
./include/linux/compiler.h:327:21: note: in expansion of macro ‘__native_word’
  compiletime_assert(__native_word(t),    \
                     ^
./arch/x86/include/asm/barrier.h:97:2: note: in expansion of macro ‘compiletime_assert_atomic_type’
  compiletime_assert_atomic_type(*p);    \
  ^
/var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/../../../../include-overrides/asm-generic/barrier.h:157:33: note: in expansion of macro ‘__smp_store_release’
 #define smp_store_release(p, v) __smp_store_release(p, v)
                                 ^
./arch/x86/include/asm/qspinlock.h:42:2: note: in expansion of macro ‘smp_store_release’
  smp_store_release(&lock->locked, 0);
  ^
[... 102 lines cut ...]
  if (mm_pud_folded(mm))
      ^
./include/linux/mm.h: In function ‘mm_inc_nr_pmds’:
./include/linux/mm.h:1713:6: error: implicit declaration of function ‘mm_pmd_folded’ [-Werror=implicit-function-declaration]
  if (mm_pmd_folded(mm))
      ^
cc1: some warnings being treated as errors
scripts/Makefile.build:330: recipe for target '/var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_driver.o' failed
make[1]: *** [/var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_driver.o] Error 1
Makefile:1747: recipe for target 'uvcvideo.ko' failed
make: *** [uvcvideo.ko] Error 2
make: Leaving directory '/usr/src/linux-headers-4.15.0-64-generic'
decades@ubuntu:~$ 
```

Also the final check for the kernel patch does not show the expected string:

```
decades@ubuntu:~$ modinfo uvcvideo | grep "version:"
version:        1.1.1
srcversion:     79A49299483C7A964CCC049

```

At runtime the RealSense Viewer complains about missing `/dev/video0`, but it works from time to time. Also today I got it running again after several attempts on Ubuntu. But it is all the time a tricky thing to make it work.

ev-mp, 2019-09-23:
@neilyoung thanks, we're are aware of the issue with DKMS with the recent releases on Ubuntu 4.15 branch - #4869, #4669. But those should not affect the presence of `dev/video*`.
If indeed, you observe intermittent reports of missing device in the `viewer` then monitoring the actual device connection with the kernel log for HW issues is the first priority task. 
Before that I would also recommend to try changing the USB port/cable.

neilyoung, 2019-09-23:
@ev-mp Thanks. The cable is the original cable. I also tried another one already to no avail. Rest of info will follow in an hour or so. The Windows process is at 50% :/

neilyoung, 2019-09-23:
@ev-mp OK, here we have it. Right now it is again not working.
The output of of the commands you mentioned (plugging and unplugging the device):

`dmesg -wT`

```
[Mon Sep 23 17:06:52 2019] usb 2-1: new SuperSpeed USB device number 4 using xhci_hcd
[Mon Sep 23 17:06:52 2019] usb 2-1: New USB device found, idVendor=8086, idProduct=0b07
[Mon Sep 23 17:06:52 2019] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Mon Sep 23 17:06:52 2019] usb 2-1: Product: Intel(R) RealSense(TM) Depth Camera 435 
[Mon Sep 23 17:06:52 2019] usb 2-1: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435 
[Mon Sep 23 17:06:52 2019] usb 2-1: SerialNumber: 935723023861
[Mon Sep 23 17:07:27 2019] usb 2-1: USB disconnect, device number 4

```

`rs-enumerate-devices`

```
decades@ubuntu:~$ rs-enumerate-devices 
No device detected. Is it plugged in?
```

ev-mp, 2019-09-23:
@neilyoung , did you manually disconnect/reset the camera at this point ?
>[Mon Sep 23 17:07:27 2019] usb 2-1: USB disconnect, device number 4

If not - it would help to connect the camera via an external self-powered USB3 hub and see whether the disconnect events still occur.

neilyoung, 2019-09-23:
@ev-mp Good point. Let me test that.

neilyoung, 2019-09-23:
@ev-mp No, doesn't work either

> @neilyoung , did you manually disconnect/reset the camera at this point ?
> 
> [Mon Sep 23 17:07:27 2019] usb 2-1: USB disconnect, device number 4
> 
> 
Yes

ev-mp, 2019-09-23:
@neilyoung , I'm not sure I understand - this mean that the disconnect was made by you, not internal event, correct ?
It could be that for some reason the UVC driver failed to load automatically - try to run it manually with `sudo modprobe uvcvideo`, and then run `rs-enumerate-devices` while monitoring the kernel log

neilyoung, 2019-09-23:
@ev-mp Right. I just plugged the device and unplugged it again. But of course while running the viewer it was plugged :)

Did the `sudo modprobe uvcvideo`. After that the result of `rs-enumerate-devices` was the same. "No device found".

I then opened a new console and run `dmesg -wT` while `rs-enumerate-devices` in another console (if that was it, what you wanted me to do)

There was no visible output in the `dmesg` window.

neilyoung, 2019-09-23:
Here is the output of `dmesg` while plugging the camera. Does it help?

```
[Mon Sep 23 17:36:52 2019] usb 2-1: USB disconnect, device number 10
[Mon Sep 23 17:36:52 2019] uvcvideo: Failed to query (GET_CUR) UVC control 7 on unit 3: -108 (exp. 1).
[Mon Sep 23 17:36:56 2019] usb 2-1: new SuperSpeed USB device number 11 using xhci_hcd
[Mon Sep 23 17:36:56 2019] usb 2-1: New USB device found, idVendor=8086, idProduct=0b07
[Mon Sep 23 17:36:56 2019] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Mon Sep 23 17:36:56 2019] usb 2-1: Product: Intel(R) RealSense(TM) Depth Camera 435 
[Mon Sep 23 17:36:56 2019] usb 2-1: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435 
[Mon Sep 23 17:36:56 2019] usb 2-1: SerialNumber: 935723023861
[Mon Sep 23 17:36:56 2019] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[Mon Sep 23 17:36:56 2019] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[Mon Sep 23 17:36:56 2019] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[Mon Sep 23 17:36:56 2019] uvcvideo: Unable to create debugfs 2-11 directory.
[Mon Sep 23 17:36:56 2019] uvcvideo 2-1:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[Mon Sep 23 17:36:56 2019] uvcvideo 2-1:1.0: Entity type for entity Processing 2 was not initialized!
[Mon Sep 23 17:36:56 2019] uvcvideo 2-1:1.0: Entity type for entity Camera 1 was not initialized!
[Mon Sep 23 17:36:56 2019] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/input/input21
[Mon Sep 23 17:36:56 2019] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[Mon Sep 23 17:36:56 2019] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[Mon Sep 23 17:36:56 2019] uvcvideo: Unable to create debugfs 2-11 directory.
[Mon Sep 23 17:36:56 2019] uvcvideo 2-1:1.3: Entity type for entity Processing 7 was not initialized!
[Mon Sep 23 17:36:56 2019] uvcvideo 2-1:1.3: Entity type for entity Extension 8 was not initialized!
[Mon Sep 23 17:36:56 2019] uvcvideo 2-1:1.3: Entity type for entity Camera 6 was not initialized!
[Mon Sep 23 17:36:56 2019] ------------[ cut here ]------------
[Mon Sep 23 17:36:56 2019] Unknown pixelformat 0x00000000
[Mon Sep 23 17:36:56 2019] WARNING: CPU: 4 PID: 5604 at /build/linux-hwe-t5DhM2/linux-hwe-4.15.0/drivers/media/v4l2-core/v4l2-ioctl.c:1293 v4l_fill_fmtdesc+0xf4d/0x1260 [videodev]
[Mon Sep 23 17:36:56 2019] Modules linked in: uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core videodev media rfcomm ipt_MASQUERADE nf_nat_masquerade_ipv4 nf_conntrack_netlink nfnetlink xfrm_user xfrm_algo iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 xt_addrtype iptable_filter ip_tables xt_conntrack x_tables nf_nat nf_conntrack libcrc32c br_netfilter bridge stp llc ccm aufs pci_stub vboxpci(OE) vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) overlay bnep nls_iso8859_1 intel_rapl x86_pkg_temp_thermal intel_powerclamp snd_soc_skl coretemp kvm_intel kvm snd_soc_skl_ipc irqbypass snd_hda_ext_core snd_soc_sst_dsp snd_soc_sst_ipc crct10dif_pclmul 8250_dw snd_soc_acpi snd_soc_core snd_compress crc32_pclmul ac97_bus ghash_clmulni_intel snd_pcm_dmaengine snd_hda_codec_hdmi hid_multitouch spi_pxa2xx_platform
[Mon Sep 23 17:36:56 2019]  btusb snd_hda_codec_generic pcbc aesni_intel snd_hda_intel snd_hda_codec joydev btrtl btbcm snd_hda_core snd_hwdep btintel bluetooth aes_x86_64 snd_pcm crypto_simd ecdh_generic snd_seq_midi arc4 glue_helper iwlmvm snd_seq_midi_event mac80211 iwlwifi cfg80211 cryptd snd_rawmidi input_leds intel_cstate snd_seq snd_seq_device snd_timer shpchp snd serio_raw wmi_bmof asus_nb_wmi intel_rapl_perf asus_wmi sparse_keymap idma64 virt_dma intel_lpss_pci intel_lpss mei_me soundcore mei processor_thermal_device int3403_thermal intel_pch_thermal int340x_thermal_zone intel_soc_dts_iosf int3400_thermal asus_wireless acpi_thermal_rel mac_hid acpi_pad cuse parport_pc ppdev lp parport autofs4 hid_generic uas usb_storage usbhid i915 i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops
[Mon Sep 23 17:36:56 2019]  ahci i2c_hid drm libahci wmi hid video
[Mon Sep 23 17:36:56 2019] CPU: 4 PID: 5604 Comm: chrome Tainted: G        W  OE    4.15.0-64-generic #73~16.04.1-Ubuntu
[Mon Sep 23 17:36:56 2019] Hardware name: ASUSTeK COMPUTER INC. X510UAR/X510UAR, BIOS X510UAR.301 09/25/2017
[Mon Sep 23 17:36:56 2019] RIP: 0010:v4l_fill_fmtdesc+0xf4d/0x1260 [videodev]
[Mon Sep 23 17:36:56 2019] RSP: 0018:ffffaedb40f3fc50 EFLAGS: 00010286
[Mon Sep 23 17:36:56 2019] RAX: 0000000000000000 RBX: ffffaedb40f3fd68 RCX: 0000000000000006
[Mon Sep 23 17:36:56 2019] RDX: 0000000000000007 RSI: 0000000000000082 RDI: ffff8d929ed16490
[Mon Sep 23 17:36:56 2019] RBP: ffffaedb40f3fc60 R08: 0000000000000001 R09: 000000000000064e
[Mon Sep 23 17:36:56 2019] R10: ffff8d91e40cb300 R11: 000000000000064e R12: ffffffffc0f40860
[Mon Sep 23 17:36:56 2019] R13: ffff8d91e40cb300 R14: ffff8d91dc16af00 R15: 0000000000000000
[Mon Sep 23 17:36:56 2019] FS:  00007f0621536b40(0000) GS:ffff8d929ed00000(0000) knlGS:0000000000000000
[Mon Sep 23 17:36:56 2019] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[Mon Sep 23 17:36:56 2019] CR2: 00005574b5df8720 CR3: 000000018918a004 CR4: 00000000003606e0
[Mon Sep 23 17:36:56 2019] Call Trace:
[Mon Sep 23 17:36:56 2019]  v4l_enum_fmt+0x7e/0x140 [videodev]
[Mon Sep 23 17:36:56 2019]  __video_do_ioctl+0x2a4/0x320 [videodev]
[... cut ...]
[Mon Sep 23 17:36:56 2019] R10: 00007f0621536b40 R11: 0000000000000246 R12: 00003b01403864c0
[Mon Sep 23 17:36:56 2019] R13: 00003b014039f558 R14: 0000000000000019 R15: 00003b01403a3210
[Mon Sep 23 17:36:56 2019]  ? uas_probe+0x4f2/0x550 [uas]
[Mon Sep 23 17:36:56 2019] Code: 30 0f 84 b4 01 00 00 3d 4a 4c 32 30 0f 84 97 01 00 00 3d 4d 33 31 30 0f 84 7a 01 00 00 89 c6 48 c7 c7 b2 10 f0 c0 e8 63 35 5a c0 <0f> 0b 80 7b 0c 00 0f 85 d6 f3 ff ff 8b 43 2c 48 c7 c1 59 04 f0 
[Mon Sep 23 17:36:56 2019] ---[ end trace fa79cb9a589837d4 ]---
```

neilyoung, 2019-09-23:
I tried to work around with a fresh live install of Ubuntu 18.04, but things are getting worse.
While installing `librealsense2-util` I'm getting an error, that `librealsense2-gl` is required, but should not be installed (?)

Giving up. Camera goes back

neilyoung, 2019-09-25:
@ev-mp Thanks, I have returned the camera. Thanks for your efforts

ev-mp, 2019-09-25:
@neilyoung , thanks for the cooperation.

closed 2019-09-25 by ev-mp

neilyoung, 2019-09-25:
@ev-mp Thank you. This is for sure not the end of the story :)

thorroed, 2020-05-27:
has there been any update regarding this issue?

We are experiencing the similar crash trace after stopping the RealSense 435i using an UP board with this setup:

- Ubuntu 18.04
- Kernel 4.15.0 [Installed according to this documentation](https://wiki.up-community.org/Ubuntu#Install_Ubuntu_kernel_4.15.0_for_UP_from_PPA_on_Ubuntu_18.04_.28Not_UP_Xtreme.29
)
- RealSense D435I firmware version: 05.12.03.00, USB type: 3.2
- librealsense2 version: 2.34.0
- Logitech BRIO camera connected to USB2 

Here is the output of dmesg -wT that we see:

```Mi Mai 27 08:06:18 2020] uvcvideo: Non-zero status (-71) in video completion handler.
[Mi Mai 27 08:06:18 2020] uvcvideo: Non-zero status (-71) in video completion handler.
[Mi Mai 27 08:06:18 2020] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[Mi Mai 27 08:06:18 2020] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[Mi Mai 27 08:06:18 2020] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[Mi Mai 27 08:06:18 2020] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[Mi Mai 27 08:06:19 2020] ------------[ cut here ]------------
[Mi Mai 27 08:06:19 2020] Unknown pixelformat 0x00000000
[Mi Mai 27 08:06:19 2020] WARNING: CPU: 3 PID: 4810 at /build/linux-hwe-KDyd1s/linux-hwe-4.15.0/drivers/media/v4l2-core/v4l2-ioctl.c:1291 v4l_fill_fmtdesc+0xf41/0x1250 [videodev]
[Mi Mai 27 08:06:19 2020] Modules linked in: xt_nat xt_tcpudp veth xt_conntrack ipt_MASQUERADE nf_nat_masquerade_ipv4 nf_conntrack_netlink nfnetlink xfrm_user xfrm_algo xt_addrtype iptable_filter iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack libcrc32c br_netfilter bridge stp llc aufs overlay cmac bnep nls_iso8859_1 snd_hda_codec_hdmi snd_soc_skl snd_soc_skl_ipc snd_hda_ext_core snd_soc_sst_dsp snd_soc_sst_ipc snd_soc_acpi snd_soc_core snd_compress ac97_bus snd_pcm_dmaengine spi_pxa2xx_platform 8250_dw leds_upboard pinctrl_upboard intel_rapl intel_telemetry_pltdrv intel_punit_ipc intel_telemetry_core intel_pmc_ipc x86_pkg_temp_thermal coretemp kvm_intel kvm irqbypass snd_hda_intel crct10dif_pclmul crc32_pclmul ghash_clmulni_intel uvcvideo snd_hda_codec snd_usb_audio pcbc snd_hda_core
[Mi Mai 27 08:06:19 2020]  videobuf2_vmalloc snd_usbmidi_lib videobuf2_memops hid_sensor_accel_3d hid_sensor_gyro_3d videobuf2_v4l2 snd_hwdep hid_sensor_trigger videobuf2_core industrialio_triggered_buffer kfifo_buf snd_pcm hid_sensor_iio_common videodev pl2303 industrialio aesni_intel btusb snd_seq_midi btrtl usbserial aes_x86_64 media crypto_simd glue_helper cryptd snd_seq_midi_event joydev snd_rawmidi btbcm input_leds btintel arc4 intel_cstate snd_seq intel_rapl_perf bluetooth i915 ecdh_generic snd_seq_device iwlmvm mac80211 iwlwifi pwm_lpss_pci lpc_ich snd_timer pwm_lpss drm_kms_helper idma64 virt_dma cfg80211 drm intel_lpss_pci intel_lpss i2c_algo_bit snd mac_hid fb_sys_fops syscopyarea video sysfillrect sysimgblt soundcore upboard_fpga mei_me mei shpchp sch_fq_codel parport_pc ppdev lp parport ip_tables
[Mi Mai 27 08:06:19 2020]  x_tables autofs4 hid_sensor_custom hid_sensor_hub hid_generic usbhid hid mmc_block dwc3 ulpi udc_core sdhci_pci sdhci dwc3_pci r8169 mii ahci libahci pinctrl_broxton
[Mi Mai 27 08:06:19 2020] CPU: 3 PID: 4810 Comm: python3 Tainted: G        W        4.15.0-37-generic #40~upboard06-Ubuntu
[Mi Mai 27 08:06:19 2020] Hardware name: AAEON UP-APL01/UP-APL01, BIOS UPA1AM21 09/01/2017
[Mi Mai 27 08:06:19 2020] RIP: 0010:v4l_fill_fmtdesc+0xf41/0x1250 [videodev]
[Mi Mai 27 08:06:19 2020] RSP: 0018:ffffa9c68434fc30 EFLAGS: 00010282
[Mi Mai 27 08:06:19 2020] RAX: 0000000000000000 RBX: ffffa9c68434fd68 RCX: 0000000000000006
[Mi Mai 27 08:06:19 2020] RDX: 0000000000000007 RSI: 0000000000000086 RDI: ffff9b98ffd96490
[Mi Mai 27 08:06:19 2020] RBP: ffffa9c68434fc40 R08: 0000000000000001 R09: 0000000000000887
[Mi Mai 27 08:06:19 2020] R10: ffff9b97e70f5c00 R11: 0000000000000000 R12: ffff9b97e7342600
[Mi Mai 27 08:06:19 2020] R13: ffffffffc0ab2900 R14: ffff9b97e70f5c00 R15: ffff9b97e7342600
[Mi Mai 27 08:06:19 2020] FS:  00007f03eb1c3740(0000) GS:ffff9b98ffd80000(0000) knlGS:0000000000000000
[Mi Mai 27 08:06:19 2020] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[Mi Mai 27 08:06:19 2020] CR2: 00000000023a07b0 CR3: 000000003f916000 CR4: 00000000003406e0
[Mi Mai 27 08:06:19 2020] Call Trace:
[Mi Mai 27 08:06:19 2020]  v4l_enum_fmt+0x74/0x110 [videodev]
[Mi Mai 27 08:06:19 2020]  __video_do_ioctl+0x370/0x380 [videodev]
[Mi Mai 27 08:06:19 2020]  video_usercopy+0x24d/0x640 [videodev]
[Mi Mai 27 08:06:19 2020]  ? video_ioctl2+0x20/0x20 [videodev]
[Mi Mai 27 08:06:19 2020]  ? futex_wake+0x8f/0x180
[Mi Mai 27 08:06:19 2020]  video_ioctl2+0x15/0x20 [videodev]
[Mi Mai 27 08:06:19 2020]  v4l2_ioctl+0xc8/0xf0 [videodev]
[Mi Mai 27 08:06:19 2020]  do_vfs_ioctl+0xa8/0x630
[Mi Mai 27 08:06:19 2020]  ? SyS_futex+0x13b/0x180
[Mi Mai 27 08:06:19 2020]  SyS_ioctl+0x79/0x90
[Mi Mai 27 08:06:19 2020]  do_syscall_64+0x73/0x130
[Mi Mai 27 08:06:19 2020]  entry_SYSCALL_64_after_hwframe+0x3d/0xa2
[Mi Mai 27 08:06:19 2020] RIP: 0033:0x7f03eacde5d7
[Mi Mai 27 08:06:19 2020] RSP: 002b:00007ffda6c901c8 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[Mi Mai 27 08:06:19 2020] RAX: ffffffffffffffda RBX: 00007ffda6c90570 RCX: 00007f03eacde5d7
[Mi Mai 27 08:06:19 2020] RDX: 00007ffda6c90370 RSI: 00000000c0405602 RDI: 000000000000000a
[... cut ...]
[Mi Mai 27 08:06:19 2020] R10: 000000000233d010 R11: 0000000000000246 R12: 0000000000000006
[Mi Mai 27 08:06:19 2020] R13: 000000005a313620 R14: 0000000000000500 R15: 00007ffda6c90770
[Mi Mai 27 08:06:19 2020] Code: 30 0f 84 b4 01 00 00 3d 4a 4c 32 30 0f 84 97 01 00 00 3d 4d 33 31 30 0f 84 7a 01 00 00 89 c6 48 c7 c7 d7 a2 9b c0 e8 bf 67 2e f5 <0f> 0b 80 7b 0c 00 0f 85 e2 f3 ff ff 8b 43 2c 48 c7 c1 7e 96 9b
[Mi Mai 27 08:06:19 2020] ---[ end trace fa79344827d6ba17 ]---```

Thank you for any help you can give us.

pirate999, 2020-07-20:
Have the problem solved? I have the same problem

mau-jebi, 2022-05-21:
Any update on this issue?
We have these cameras (D455 & L515) on a Computer Vision System and suddenly they are not recognized anymore. Rebooting the computer (Jetson Xavier AGX) does not solve the issue. 

@INTEL: Please take a look at this issue, it really renders these devices unreliable for serious applications. 

Our setup:
NVIDIA Jetson 
Ubuntu 18.04
Kernel 4.9.201 
RealSense D455 & L515 (Both connected to the system)
firmware version: Not available since the camera is not available.
LibreSense SDK: 2.47
USB type: 3.0

Here is a portion of the output of dmesg -wT that we see:

[Sat May 21 07:47:21 2022] zram5: detected capacity change from 0 to 2092478464
[Sat May 21 07:47:21 2022] Adding 2043432k swap on /dev/zram5.  Priority:5 extents:1 across:2043432k SS
[Sat May 21 07:47:21 2022] zram6: detected capacity change from 0 to 2092478464
[Sat May 21 07:47:21 2022] Adding 2043432k swap on /dev/zram6.  Priority:5 extents:1 across:2043432k SS
[Sat May 21 07:47:21 2022] zram7: detected capacity change from 0 to 2092478464
[Sat May 21 07:47:21 2022] Adding 2043432k swap on /dev/zram7.  Priority:5 extents:1 across:2043432k SS
[Sat May 21 07:47:21 2022] Bridge firewalling registered
[Sat May 21 07:47:21 2022] nf_conntrack version 0.5.0 (65536 buckets, 262144 max)
[Sat May 21 07:47:21 2022] Netfilter messages via NETLINK v0.30.
[Sat May 21 07:47:21 2022] ctnetlink v0.93: registering with nfnetlink.
[Sat May 21 07:47:22 2022] IPv6: ADDRCONF(NETDEV_UP): docker0: link is not ready
[Sat May 21 07:47:22 2022] usb 2-3: Device not responding to setup address.
[Sat May 21 07:47:22 2022] tegradc 15200000.nvdisplay: blank - powerdown
[Sat May 21 07:47:22 2022] tegradc 15220000.nvdisplay: blank - powerdown
[Sat May 21 07:47:22 2022] tegradc 15210000.nvdisplay: blank - powerdown
[Sat May 21 07:47:22 2022] usb 2-3: device not accepting address 2, error -71
[Sat May 21 07:47:22 2022] usb 1-3: new full-speed USB device number 3 using tegra-xusb
[Sat May 21 07:47:22 2022] tegra-xusb 3610000.xhci: Error: Failed finding new dequeue state
[Sat May 21 07:47:22 2022] tegra-xusb 3610000.xhci: controller firmware hang
[Sat May 21 07:47:22 2022] tegra-xusb 3610000.xhci: hcd_reinit is disabled or in progress
[Sat May 21 07:47:25 2022] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[Sat May 21 07:47:25 2022] Bluetooth: BNEP socket layer initialized
[Sat May 21 07:47:26 2022] usb usb2-port3: Cannot enable. Maybe the USB cable is bad?
[Sat May 21 07:47:26 2022] usb usb2-port3: attempt power cycle
[Sat May 21 07:47:35 2022] tegra-xusb 3610000.xhci: Stopped the command ring failed, maybe the host is dead
[Sat May 21 07:47:35 2022] tegra-xusb 3610000.xhci: Host not halted after 16000 microseconds.
[Sat May 21 07:47:35 2022] tegra-xusb 3610000.xhci: Abort command ring failed
[Sat May 21 07:47:35 2022] bpmp: mrq 22 took 3984000 us
[Sat May 21 07:47:35 2022] tegra-xusb 3610000.xhci: xHCI host not responding to stop endpoint command.
[Sat May 21 07:47:35 2022] tegra-xusb 3610000.xhci: Assuming host is dying, halting host.
[Sat May 21 07:47:35 2022] tegra-xusb 3610000.xhci: Host not halted after 16000 microseconds.
[Sat May 21 07:47:35 2022] tegra-xusb 3610000.xhci: Non-responsive xHCI host is not halting.
[Sat May 21 07:47:35 2022] tegra-xusb 3610000.xhci: Completing active URBs anyway.
[Sat May 21 07:47:35 2022] tegra-xusb 3610000.xhci: HC died; cleaning up
[Sat May 21 07:47:35 2022] tegra-xusb 3610000.xhci: hcd_reinit is disabled or in progress
[Sat May 21 07:47:35 2022] tegra-xusb 3610000.xhci: HC died; cleaning up
[Sat May 21 07:47:35 2022] tegra-xusb 3610000.xhci: hcd_reinit is disabled or in progress
[Sat May 21 07:47:35 2022] eqos 2490000.ether_qos eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[Sat May 21 07:47:35 2022] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[Sat May 21 07:47:36 2022] usb usb2-port3: couldn't allocate usb_device
[Sat May 21 07:47:48 2022] vdd-1v8-cvb: disabling
[Sat May 21 07:47:48 2022] vdd-1v8-sd: disabling
[Sat May 21 07:47:48 2022] vdd-epb-1v0: disabling
[Sat May 21 07:47:48 2022] avdd-cam-2v8: disabling
[Sat May 21 07:47:48 2022] vdd-sata-1v5: disabling
[Sat May 21 07:47:48 2022] vdd-1v8-slt: disabling
[Sat May 21 07:47:48 2022] vdd-3v3-slt: disabling
[Sat May 21 07:47:48 2022] vdd-fan: disabling
[Sat May 21 07:47:48 2022] vdd_sys_en: disabling
[Sat May 21 07:48:19 2022] iwlwifi 0003:01:00.0: Direct firmware load for iwlwifi-cc-a0-55.ucode failed with error -2
[Sat May 21 07:48:19 2022] iwlwifi 0003:01:00.0: Falling back to user helper
[Sat May 21 07:48:19 2022] iwlwifi 0003:01:00.0: Direct firmware load for iwlwifi-cc-a0-54.ucode failed with error -2
[Sat May 21 07:48:19 2022] iwlwifi 0003:01:00.0: Falling back to user helper
[Sat May 21 07:48:19 2022] iwlwifi 0003:01:00.0: Direct firmware load for iwlwifi-cc-a0-53.ucode failed with error -2
[Sat May 21 07:48:19 2022] iwlwifi 0003:01:00.0: Falling back to user helper
[Sat May 21 07:48:19 2022] iwlwifi 0003:01:00.0: Direct firmware load for iwlwifi-cc-a0-52.ucode failed with error -2
[Sat May 21 07:48:19 2022] iwlwifi 0003:01:00.0: Falling back to user helper
[Sat May 21 07:48:19 2022] iwlwifi 0003:01:00.0: Direct firmware load for iwlwifi-cc-a0-51.ucode failed with error -2
[Sat May 21 07:48:19 2022] iwlwifi 0003:01:00.0: Falling back to user helper
[Sat May 21 07:48:19 2022] iwlwifi 0003:01:00.0: Direct firmware load for iwlwifi-cc-a0-50.ucode failed with error -2
[... cut ...]
[Sat May 21 07:48:19 2022] iwlwifi 0003:01:00.0: Falling back to user helper
[Sat May 21 07:48:19 2022] iwlwifi 0003:01:00.0: TLV_FW_FSEQ_VERSION: FSEQ Version: 43.2.23.17
[Sat May 21 07:48:19 2022] iwlwifi 0003:01:00.0: Found debug destination: EXTERNAL_DRAM
[Sat May 21 07:48:19 2022] iwlwifi 0003:01:00.0: Found debug configuration: 0

Here is the complete output of dmesg -wT that we see:

[jebi_RS_May_21_2022.docx](https://github.com/IntelRealSense/librealsense/files/8747639/jebi_RS_May_21_2022.docx)

ciandonovan, 2023-07-05:
Issue occurs on the lastest stable Debian release, but doesn't seem to actually affect the RealSense's functionality. Below is the `dmesg` log that gets printed when the device is just first plugged in, but it doesn't complain otherwise or later on. No userspace programs are running when this occurs either so not an issue with the SDK.

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Firmware Version                   | 5.13.0.50 | 
| Operating System & Version |   Debian 12 Bookworm  | 
| Kernel Version (Linux Only)    |  6.1.0-9-amd64                                         | 
| Platform                                 | PC  |
```
[  406.101147] ------------[ cut here ]------------
[  406.104613] Unknown pixelformat 0x00000000
[  406.108186] WARNING: CPU: 4 PID: 659 at drivers/media/v4l2-core/v4l2-ioctl.c:1503 v4l_enum_fmt+0x11e2/0x1b70 [videodev]
[  406.111841] Modules linked in: uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev mc tls xt_MASQUERADE xt_mark nft_compat ctr ccm snd_seq_dummy snd_hrtimer snd_seq snd_seq_device tun amdgpu snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio snd_hda_codec_hdmi snd_hda_intel gpu_sched snd_intel_dspcfg drm_buddy snd_intel_sdw_acpi drm_display_helper snd_hda_codec hid_multitouch snd_hda_core cec hid_generic snd_pci_acp6x snd_hwdep rc_core cdc_ether drm_ttm_helper joydev snd_pci_acp5x usbhid usbnet option snd_pcm ttm snd_rn_pci_acp3x ftdi_sio usb_wwan xpad snd_acp_config snd_timer mii drm_kms_helper hid usbserial cdc_acm ff_memless snd snd_soc_acpi i2c_algo_bit soundcore snd_pci_acp3x overlay binfmt_misc intel_rapl_msr intel_rapl_common edac_mce_amd nls_ascii nls_cp437 vfat fat kvm_amd ccp rng_core kvm irqbypass ghash_clmulni_intel mt76x2u sha512_ssse3 mt76x2_common mt76x02_usb mt76_usb mt76x02_lib bridge mt76 stp sha512_generic llc mac80211
[  406.111923]  aesni_intel crypto_simd libarc4 cryptd sp5100_tco apex(OE) rapl gasket(OE) k10temp watchdog pcspkr button acpi_cpufreq cfg80211 evdev sg rfkill nft_ct nft_masq nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables libcrc32c pkcs8_key_parser nfnetlink drm fuse loop efi_pstore configfs efivarfs ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 crc32c_generic dm_mod sd_mod t10_pi crc64_rocksoft crc64 crc_t10dif crct10dif_generic ahci libahci libata xhci_pci xhci_hcd r8169 crct10dif_pclmul usbcore realtek crct10dif_common scsi_mod crc32_pclmul mdio_devres crc32c_intel scsi_common usb_common libphy i2c_piix4 i2c_amd_mp2_pci video wmi
[  406.156715] CPU: 4 PID: 659 Comm: pipewire Tainted: G           OE      6.1.0-9-amd64 #1  Debian 6.1.27-1
[  406.159858] Hardware name: To Be Filled By O.E.M. To Be Filled By O.E.M./4X4-V1000, BIOS P1.60L 05/29/2023
[  406.163170] RIP: 0010:v4l_enum_fmt+0x11e2/0x1b70 [videodev]
[  406.166381] Code: c7 c3 2b 33 c7 c1 81 fe 64 76 73 64 0f 84 53 f0 ff ff 41 80 7c 24 0c 00 0f 85 12 ef ff ff 48 c7 c7 7e 41 c7 c1 e8 ae 61 44 f6 <0f> 0b 48 c7 c2 9a 41 c7 c1 49 8d 4c 24 2c 49 8d 7c 24 0c 31 ed be
[  406.172875] RSP: 0018:ffffb1d580a17cd8 EFLAGS: 00010286
[  406.176185] RAX: 0000000000000000 RBX: ffffffffc1c738e6 RCX: 0000000000000000
[  406.179378] RDX: 0000000000000002 RSI: ffffffffb933fa66 RDI: 00000000ffffffff
[  406.182627] RBP: 0000000000000000 R08: 0000000000000000 R09: ffffb1d580a17b48
[  406.185772] R10: 0000000000000003 R11: ffffffffb9ad43a8 R12: ffffb1d580a17de8
[  406.188986] R13: 0000000000000000 R14: ffff9e080eff1d80 R15: ffff9e08328a8018
[  406.192051] FS:  00007fbe48cd0740(0000) GS:ffff9e08b7300000(0000) knlGS:0000000000000000
[  406.195167] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  406.198404] CR2: 00007fbe474ff280 CR3: 00000001192f8000 CR4: 00000000003506e0
[  406.201535] Call Trace:
[  406.204639]  <TASK>
[  406.207633]  __video_do_ioctl+0x1c0/0x410 [videodev]
[  406.210827]  video_usercopy+0x1a5/0x770 [videodev]
[  406.213776]  ? v4l_print_control+0x20/0x20 [videodev]
[  406.216801]  v4l2_ioctl+0x49/0x50 [videodev]
[  406.219814]  __x64_sys_ioctl+0x90/0xd0
[  406.222820]  do_syscall_64+0x5b/0xc0
[  406.225757]  ? syscall_exit_to_user_mode+0x17/0x40
[  406.228644]  ? do_syscall_64+0x67/0xc0
[  406.232060]  ? exit_to_user_mode_prepare+0x40/0x1d0
[  406.235165]  ? syscall_exit_to_user_mode+0x17/0x40
[  406.237975]  ? do_syscall_64+0x67/0xc0
[  406.240835]  entry_SYSCALL_64_after_hwframe+0x63/0xcd
[  406.243621] RIP: 0033:0x7fbe48dd0afb
[  406.246605] Code: 00 48 89 44 24 18 31 c0 48 8d 44 24 60 c7 04 24 10 00 00 00 48 89 44 24 08 48 8d 44 24 20 48 89 44 24 10 b8 10 00 00 00 0f 05 <89> c2 3d 00 f0 ff ff 77 1c 48 8b 44 24 18 64 48 2b 04 25 28 00 00
[  406.252140] RSP: 002b:00007ffcf07cb970 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[  406.254994] RAX: ffffffffffffffda RBX: 000000000000002c RCX: 00007fbe48dd0afb
[  406.257811] RDX: 000055d10ec2febc RSI: ffffffffc0405602 RDI: 000000000000002c
[  406.260534] RBP: 000055d10ec2ff00 R08: 00007fbe48ea5cc0 R09: 000000000000000b
[  406.263811] R10: 0000000000000000 R11: 0000000000000246 R12: 000055d10ec2febc
[  406.266631] R13: 000055d10ec2fe90 R14: 0000000000000000 R15: 000055d10ec2fcc8
[  406.269652]  ? wmi_dev_probe.cold+0x3ac/0xdaa [wmi]
[  406.272238]  </TASK>
[  406.275223] ---[ end trace 0000000000000000 ]---
[  406.278220] ------------[ cut here ]------------
[... cut ...]
[  406.828737] RAX: ffffffffffffffda RBX: 000000000000002c RCX: 00007fbe48dd0afb
[  406.832813] RDX: 000055d10ec39b6c RSI: ffffffffc0405602 RDI: 000000000000002c
[  406.836649] RBP: 00007fbe46631640 R08: 0000000000000000 R09: 0000000000000002
[  406.840958] R10: 0000000000000000 R11: 0000000000000246 R12: 000055d10ec39b6c
[  406.845002] R13: 000055d10ec3b530 R14: 0000000000000000 R15: 000055d10ec39978
[  406.849084]  ? wmi_dev_probe.cold+0x3ac/0xdaa [wmi]
[  406.852936]  </TASK>
[  406.856857] ---[ end trace 0000000000000000 ]---
```

MartyG-RealSense, 2023-07-05:
Hi @ciandonovan  The RealSense SDK does not yet officially support kernel 6.1.  At the time of writing this, the newest supported kernel version is 5.15.  It is possible to use unsupported kernels with the SDK without applying a kernel patch but there may be unforeseen consequences in regards to stability.

This issue can be resolved if the SDK is built from source code with CMake with the build flag **-DFORCE_RSUSB_BACKEND=true** included in the CMake build instruction.  An SDK build based on the RSUSB Backend bypasses the kernel, so it is not dependent on Linux versions or kernel versions and does not require kernel patching.

ciandonovan, 2023-07-06:
@MartyG-RealSense just it clarify, this issue occurs *without* the RealSense SDK, I've tested it with a bare kernel, nothing else RealSense related installed. However the RealSense firmware is presenting itself the kernel doesn't like, specifically the V4L2 component. But again, nothing to do with the RealSense SDK as it is not installed.

MartyG-RealSense, 2023-07-06:
When building librealsense from source code with the native / V4L2 backend method, the kernel patch is typically applied before the SDK is installed, and the patch modifies the kernel without the SDK being involved.  I note your statement above that your kernel is unpatched though.

Have you tried using the **lsusb** command in the Ubuntu terminal to detect the camera on the USB port?  If it is detected then the lsusb log should contain a line such as `Bus 001 Device 008: ID 8086:0b3a Intel Corp.`

ciandonovan, 2023-07-06:
@MartyG-RealSense the RealSense cameras do actually function perfectly correctly on our system, it's just that particular kernel trace that looked concerning to me. It only occurs once when the camera is plugged in however. 

We actually run the entire userspace component of the RealSense SDK in unprivileged user namespace containers, using a patch I wrote to librealsense that's awaiting a Pull Request into upstream. So definitely no kernel patches, modules, or other RealSense SDK components on the host system due to that sandboxing, but everything is still functional regardless.

https://github.com/IntelRealSense/librealsense/pull/11900

Anyway I'll flag this issue with the Debian kernel team, might be under their remit.

ciandonovan, 2023-07-06:
https://github.com/IntelRealSense/meta-intel-realsense/blob/master/recipes-kernel/linux/files/6.1-realsense-camera-formats.patch

This looks like the relevant patch addressing the camera formats in v4l2-ioctl.c for the RealSense, I'll see if it resolves the issue.

MartyG-RealSense, 2023-07-06:
Thanks very much for the further details.  Please do update me when you have news.