realsenseai/librealsense #4917: modprobe: ERROR: could not insert 'uvcvideo': Invalid argument on Ubuntu 14.04.6 LTS

ravijo, 2019-09-23:
| Required Info                |                      |
|------------------------------|--------------------- |
| Camera Model                 | D435                 | 
| Firmware Version             |                      | 
| Operating System & Version   |  Ubuntu 14.04.6 LTS  | 
| Kernel Version (Linux Only)  |  4.4.0-148-generic   | 
| Platform                     | PC                   |
| SDK Version                  |                      | 
| Language                     |                      | 
| Segment                      |                      | 

### Issue Description
The process of applying a patch on kernel modules is failing. Please note that I have already run `sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade` and updated boot by `sudo update-grub && sudo reboot` multiple times. Please see below the output reported at the terminal-

```
ravi@lab:~/tools/librealsense$ ./scripts/patch-realsense-ubuntu-lts.sh

Reading package lists... 0%
Reading package lists... 100%
Reading package lists... Done

Building dependency tree... 0%
Building dependency tree... 50%
Building dependency tree

Reading state information... 0%
Reading state information... Done

bc is already the newest version.
build-essential is already the newest version.
git is already the newest version.
linux-headers-generic is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Create patches workspace in ubuntu-xenial-master folder

The kernel has modified files:
	modified:   drivers/iio/accel/hid-sensor-accel-3d.c
	modified:   drivers/iio/gyro/hid-sensor-gyro-3d.c
	modified:   drivers/media/usb/uvc/Makefile
	modified:   drivers/media/usb/uvc/uvc_ctrl.c
	modified:   drivers/media/usb/uvc/uvc_driver.c
	modified:   drivers/media/usb/uvc/uvc_video.c
	modified:   drivers/media/usb/uvc/uvcvideo.h
	modified:   drivers/media/v4l2-core/v4l2-ioctl.c
	modified:   include/linux/hid-sensor-ids.h
	modified:   include/uapi/linux/videodev2.h
Proceeding will reset all local kernel changes. Press 'n' within 3 seconds to abort the operation
Do you want to proceed? [Y/n]Y
Update the folder content with the latest from mainline branch

remote: Total 0 (delta 0), reused 0 (delta 0)
From https://git.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux/+git/xenial
 * branch            master     -> FETCH_HEAD
Resetting local changes in ubuntu-xenial-master folder
 HEAD is now at 1296204 UBUNTU: Ubuntu-4.4.0-164.192
Applying patches for xenial-master line
Applying realsense-uvc patch
patching file drivers/media/usb/uvc/Makefile
patching file drivers/media/usb/uvc/uvc_driver.c
patching file drivers/media/usb/uvc/uvcvideo.h
patching file drivers/media/v4l2-core/v4l2-ioctl.c
patching file include/uapi/linux/videodev2.h
Applying realsense-metadata patch
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 2768 (offset 69 lines).
patching file drivers/media/usb/uvc/uvc_video.c
Hunk #1 succeeded at 1253 (offset 25 lines).
Hunk #2 succeeded at 1291 (offset 25 lines).
[... 388 lines cut ...]
    [ 2330.074632] v4l2_common: disagrees about version of symbol v4l2_subdev_init
    [ 2330.074639] v4l2_common: Unknown symbol v4l2_subdev_init (err -22)
    [ 2330.074647] v4l2_common: disagrees about version of symbol v4l2_device_register_subdev
    [ 2330.074650] v4l2_common: Unknown symbol v4l2_device_register_subdev (err -22)
    [ 2358.856357] usb 2-4: new SuperSpeed USB device number 4 using xhci_hcd
    [ 2358.873291] usb 2-4: New USB device found, idVendor=8086, idProduct=0b07
    [ 2358.873300] usb 2-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
    [ 2358.873306] usb 2-4: Product: Intel(R) RealSense(TM) Depth Camera 435 
    [ 2358.873312] usb 2-4: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435 
    [ 2358.873317] usb 2-4: SerialNumber: 816513022023
    [ 2358.906688] v4l2_common: disagrees about version of symbol v4l2_subdev_init
    [ 2358.906698] v4l2_common: Unknown symbol v4l2_subdev_init (err -22)
    [ 2358.906709] v4l2_common: disagrees about version of symbol v4l2_device_register_subdev
    [ 2358.906714] v4l2_common: Unknown symbol v4l2_device_register_subdev (err -22)
    [ 2358.916355] v4l2_common: disagrees about version of symbol v4l2_subdev_init
    [ 2358.916356] v4l2_common: Unknown symbol v4l2_subdev_init (err -22)
    [ 2358.916359] v4l2_common: disagrees about version of symbol v4l2_device_register_subdev
    [ 2358.916360] v4l2_common: Unknown symbol v4l2_device_register_subdev (err -22)
    [ 2365.210215] v4l2_common: disagrees about version of symbol v4l2_subdev_init
    [ 2365.210218] v4l2_common: Unknown symbol v4l2_subdev_init (err -22)
    [ 2365.210220] v4l2_common: disagrees about version of symbol v4l2_device_register_subdev
    [ 2365.210221] v4l2_common: Unknown symbol v4l2_device_register_subdev (err -22)
    ```

Any workaround, please?

ev-mp, 2019-09-23:
@ravijo hello,
After updating the system you also need to specify the kernel to boot with - it is possible that it is predefined with the system you run to select 4.4.0-148. Using a 3rd-party tool, such as `grub-customizer` can be useful to manually pre-select the boot kernel version.

But before doing that - we recommend to use DKMS package whenever applicable as this a robust industrial solution compared to patching the kernel manually. 
Follow the [Debian installation](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) and run
>sudo apt install librealsense2-dkms

This will deploy and install both the kernel patches and the `udev-rules` required by the SDK.

ravijo, 2019-09-23:
Hi @ev-mp

Thanks for the quick response.

> After updating the system you also need to specify the kernel to boot with - it is possible that it is predefined with the system you run to select 4.4.0-148. Using a 3rd-party tool, such as `grub-customizer` can be useful to manually pre-select the boot kernel version.

I understand. Well, the kernel before was 3.13.0-165-generic. As per the pre-requirements for librealsense, I upgraded the kernel to 4.4.0-148 by using command line. *(This time I didn’t use `grub-customizer`).* However, I agree with you that I didn’t manually select the kernel during boot process. I believe it automatically executes the latest version. That is why `uname -r` is showing `4.4.0-148-generic`. Isn’t it? 

> But before doing that - we recommend to use DKMS package whenever applicable as this a robust industrial solution compared to patching the kernel manually.

I tried it in the first place. Unfortunately, while registering the server's public key, it shows timed out. It is probably due to the firewall. I am so sorry that I don’t have credentials for bypassing this firewall (this is why I changed `git://` to `https://` in the patch script as mentioned above).

Thanks again.

ev-mp, 2019-09-23:
@ravijo hello, using proxy does not require credential, just getting the corporate proxy server url:port should suffice. There are additional method to handle firewall-related issues - see the [troubleshooting link](https://unix.stackexchange.com/questions/361213/unable-to-add-gpg-key-with-apt-key-behind-a-proxy) mentioned in the installation guide.

I will provide a direct download link for convenience -  http://realsense-hw-public.s3-eu-west-1.amazonaws.com/Debian/apt-repo/pool/xenial/main/librealsense2-dkms_1.3.5-0ubuntu1_all.deb

Note that this is for `xenial` only. Install with `sudo dpkg -i <filename>` and update

ravijo, 2019-09-23:
Hi @ev-mp 

The provided `.deb` file worked like a charm. I will check the firewall stuff next time. I am closing this issue for now. Thanks!

ravijo, 2019-09-23:
Hi @ev-mp 

The installation worked. Please see below-
```
ravi@lab:~/Desktop$ sudo dpkg -i librealsense2-dkms_1.3.5-0ubuntu1_all.deb 
[sudo] password for ravi: 
Selecting previously unselected package librealsense2-dkms.
(Reading database ... 572225 files and directories currently installed.)
Preparing to unpack librealsense2-dkms_1.3.5-0ubuntu1_all.deb ...
Unpacking librealsense2-dkms (1.3.5-0ubuntu1) ...
Setting up librealsense2-dkms (1.3.5-0ubuntu1) ...
Loading new librealsense2-dkms-1.3.5 DKMS files...
First Installation: checking all kernels...
Building only for 4.4.0-148-generic
Building initial module for 4.4.0-148-generic
Done.

uvcvideo:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.4.0-148-generic/updates/dkms/

videodev.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.4.0-148-generic/updates/dkms/

hid-sensor-accel-3d.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.4.0-148-generic/updates/dkms/

hid-sensor-gyro-3d.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.4.0-148-generic/updates/dkms/

depmod....

DKMS: install completed.

Loading the modified modules into kernel... complete
Current status:
bbswitch, 0.7, 3.13.0-164-generic, x86_64: installed
bbswitch, 0.7, 3.13.0-165-generic, x86_64: installed
bbswitch, 0.7, 3.13.0-170-generic, x86_64: installed
bbswitch, 0.7, 4.2.0-42-generic, x86_64: installed
bbswitch, 0.7, 4.4.0-148-generic, x86_64: installed
librealsense2-dkms, 1.3.5, 4.4.0-148-generic, x86_64: installed
nvidia-384, 384.130, 3.13.0-164-generic, x86_64: installed
nvidia-384, 384.130, 3.13.0-165-generic, x86_64: installed
nvidia-384, 384.130, 3.13.0-170-generic, x86_64: installed
nvidia-384, 384.130, 4.2.0-42-generic, x86_64: installed
nvidia-384, 384.130, 4.4.0-148-generic, x86_64: installed
mod: videodev 			 version: realsense2-dkms
mod: uvcvideo 			 version: 1.1.2.realsense-1.3.5
mod: hid_sensor_gyro_3d 	 version: realsense2-dkms
mod: hid_sensor_accel_3d 	 version: realsense2-dkms
```

However, `realsense-viewer` is not working as expected. Please see below-
```
ravi@lab:~/Desktop$ realsense-viewer 
 23/09 22:35:47,023 ERROR [140076101125888] (types.h:304) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 23/09 22:35:47,734 ERROR [140076117911296] (types.h:304) map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
 23/09 22:35:47,734 ERROR [140076117911296] (error-handling.cpp:68) Error during polling error handler: map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
[... 89 lines cut ...]
 23/09 22:36:30,747 ERROR [140076117911296] (types.h:304) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 23/09 22:36:30,747 ERROR [140076117911296] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 23/09 22:36:31,747 ERROR [140076117911296] (types.h:304) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 23/09 22:36:31,747 ERROR [140076117911296] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 23/09 22:36:32,747 ERROR [140076117911296] (types.h:304) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 23/09 22:36:32,747 ERROR [140076117911296] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 23/09 22:36:33,748 ERROR [140076117911296] (types.h:304) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 23/09 22:36:33,748 ERROR [140076117911296] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 23/09 22:36:34,748 ERROR [140076117911296] (types.h:304) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 23/09 22:36:34,748 ERROR [140076117911296] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 23/09 22:36:35,748 ERROR [140076117911296] (types.h:304) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 23/09 22:36:35,748 ERROR [140076117911296] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 23/09 22:36:36,749 ERROR [140076117911296] (types.h:304) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 23/09 22:36:36,749 ERROR [140076117911296] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 23/09 22:36:37,749 ERROR [140076117911296] (types.h:304) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 23/09 22:36:37,749 ERROR [140076117911296] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 23/09 22:36:38,749 ERROR [140076117911296] (types.h:304) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
[... cut ...]
 23/09 22:36:39,750 ERROR [140076117911296] (types.h:304) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
 23/09 22:36:39,750 ERROR [140076117911296] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Bad file descriptor
```

If you notice carefully, you can see the following error `Cannot open '/dev/video0 Last Error: Permission denied`

Do I need to manually set permissions to the device? Can you please help?

ev-mp, 2019-09-23:
@ravijo , you need to install device permissions - `udev-rules` - follow the [installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) here
>Run Intel Realsense permissions script located from librealsense root directory:
./scripts/setup_udev_rules.sh

Then reconnect the camera to make sure the're applied

ravijo, 2019-09-23:
@ev-mp It worked. Thank you very much.

closed 2019-09-23 by ravijo