realsenseai/librealsense #4918: Intel RealSense Viewer does not recognize USB 3.0 on NVIDIA XT2

BatFresh, 2019-09-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   |0.5.11.01.100 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    | 4.4.38+                                         | 
| Platform                                 |  NVIDIA Jetson |
| SDK Version                            |  2.13.0                          | 
| Language                            |  opencv/pcl/python                         | 
| Segment			|  others                           | 

### Issue Description
Open Intel RealSense Viewer v2.13.0, Intel RealSense USB recognizes the interface of USB2, but MANIFOLD2-G is the interface of USB3.0.And click to the 3D perspective, the software will flash back

dorodnic, 2019-09-24:
2.13 was released over a year ago, in June. D435i first appears in 2.16.1 release, but I strongly urge you to use the latest. Ideally, you should follow the steps in [distribution_linux.md](https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md) to get latest precompiled version

RealSenseCustomerSupport, 2019-10-11:
----------------------------------------------

Hi @BatFresh,

Did @dorodnic's answer solve your issue?

RealSenseCustomerSupport, 2019-10-22:
----------------------------------------------

Hi @BatFresh,

Do you still need help with this question?

closed 2019-10-29 by RealSenseCustomerSupport