realsenseai/librealsense #4920: How to improve depth accuracy with multiple RealSense ?

ymatsumotomatsu, 2019-09-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i  | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version | Win 10| 
| Platform                                 | PC  |
| SDK Version                            | 2.27.0                          | 
| Language                            |  C/python                          | 
| Segment			|  AR                         | 

Hello.

I'm working on improving depth accuracy of realsense, and I found this site.
(https://forums.intel.com/s/question/0D50P0000490XJeSAM/extracts-from-the-webinar-handson-with-multiple-intel-realsense-d415-amp-d435-camera-systems?language=en_US)

In this site, 
> Accuracy can be increased by using multiple cameras instead of just one.

I already have two d435i, but I don't know how to achieve this. Please tell me how to do this.
Thanks in advance!

MartyG-RealSense, 2019-09-24:
You may find the tutorial in the link below useful for combining two or more cameras using ROS.

https://www.intelrealsense.com/how-to-multiple-camera-setup-with-ros/

ymatsumotomatsu, 2019-09-25:
@MartyG-RealSense 
Thank you for advice! I tried this way to combining pointclouds before, but I want to know how to improve depth accuracy. Do you have any good idea(e.g. function, sensor fusion, etc)?

MartyG-RealSense, 2019-09-25:
My recollection is that if you have two cameras or more in the scene and have the IR Projector enabled, the increased number of projected dots in the scene from the multiple camera projectors should in itself improve the ability to read depth information in the scene without having to do anything else.

Intel's camera tuning guide is also an excellent reference for improving the quality of images.

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf

RealSenseCustomerSupport, 2019-10-11:
----------------------------------------------

@ymatsumotomatsu, please let us know if  you need further help on this topic. Thank you.

closed 2019-10-18 by RealSenseCustomerSupport