realsenseai/librealsense #4924: Installation fails when compiling RealSense environment on NVIDIA XT2

liu978079474, 2019-09-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 0.5.11.01.100 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.4.38+                                         | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |                           | 
| Language                            |  opencv/pcl/python }                          | 
| Segment			|  others                          | 

### Issue Description
The latest way to install librealsense on NVIDIA Tx2

ev-mp, 2019-09-24:
Hi @liu978079474, you can install Librealsense2 on JetsonTX2 using Debian distribution following the [installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages).

The precompiled packages are optimized for JetsonTX2 and generated with FORCE_LIBUVC and BUILD_WITH_CUDA flags turned on.
Note that `librealsense2-dkms` module will not be available for ARM64.

You can also post the build log for review when building from sources.

liu978079474, 2019-09-28:
> Hi @liu978079474, you can install Librealsense2 on JetsonTX2 using Debian distribution following the [installation [...]

Hello,I am running sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main" -u when installing librealsense, nothing to be showed

liu978079474, 2019-09-28:
Hello,I am running sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main" -u when installing librealsense ,it show Some index files failed to download. They have been ignored, or old ones used instead.

jungladicitta, 2019-10-03:
Hello @ev-mp, should i run `./scripts/patch-realsense-ubuntu-lts.sh` on Jetson TX2?

ev-mp, 2019-10-03:
@liu978079474 , cah you share the error you get with PPA registering? Did you add the Public key first?
@jungladicitta hello, the script you mention is not applicable for Jetson TX2. Currently we provide kernel patches for Ubuntu LTS kernels on x86/x64 platforms. 
Depends on the goal you're trying to achieve you can either follow the `libuvc` path elaborated above that will deliver metadata without the need of kernel patching.
There is also a [repo](https://github.com/jetsonhacks/buildLibrealsense2TX) on JetsonHacks git dedicated to rebuild Jetson's kernel for Librealsense. But I have no info on its current status.

liu978079474, 2019-10-03:
@ev-mp Yes,I had add the Public key first.It show Some index files failed to download. They have been ignored, or old ones used instead.And I run RealSense example,something report error.

ev-mp, 2019-10-03:
@liu978079474 , can you post the full log for review?

jungladicitta, 2019-10-03:
@ev-mp thanks for the quick reply

yes, I have tried JetsonHacks scripts. Do you know how am I supposed to run ./scripts/patchKernel.sh there? I dont have kernel in my /usr/src/kernel directory

liu978079474, 2019-10-06:
@ev-mp I run sudo apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE. It show 
Executing: /tmp/tmp.tVdmQFz2kR/gpg.1.sh --keyserver
keys.gnupg.net
--recv-key
F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
gpg: requesting key 6F3EFCDE from hkp server keys.gnupg.net
gpg: key 6F3EFCDE: ""CN = Intel(R) Intel(R) Realsense", O=Intel Corporation" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1

ev-mp, 2019-10-06:
@liu978079474, the key is properly installed. So there should be no issue of registering the PPA.
Can you share the log when you try to `add-apt-repository` ?

ev-mp, 2019-10-22:
@liu978079474, in case you still experience issues with the installation then please provide the requested logs. We'll close it if no response is provided in a timely manner.

closed 2019-10-29 by RealSenseCustomerSupport