realsenseai/librealsense #4926: T265 Pose Frame Timestamp Domain

JBBee, 2019-09-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.1.0.279 | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.28.0                          | 
| Language                            |  C++                          | 
| Segment			|  AR                         | 

### Issue Description
I'm trying to synchronize the T265 pose stream with the color stream of a D435i using their frame timestamps, but I'm having a problem getting the timestamps from the same domain for comparison without having to run  [realsense_metadata_win10.ps1](https://github.com/IntelRealSense/librealsense/blob/master/scripts/realsense_metadata_win10.ps1), as our users won't be able to run this script.

Without metadata available, confirmed by running [realsense_metadata_win10.ps1](https://github.com/IntelRealSense/librealsense/blob/master/scripts/realsense_metadata_win10.ps1) `-op remove` with the two devices connected, the D435i color stream is in the `RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME` domain and the T265 pose stream is in the `RS2_TIMESTAMP_DOMAIN_GLOBAL_TIME` domain.  

If I do run the powershell script they are both in `RS2_TIMESTAMP_DOMAIN_GLOBAL_TIME` and I can compare them.  But our users won't have administrator access to their machines to be able to run this script.  

Is there a method to get the T265 stream timestamps in `RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME` so I can compare it to the D435i color stream without running  [realsense_metadata_win10.ps1](https://github.com/IntelRealSense/librealsense/blob/master/scripts/realsense_metadata_win10.ps1) as an administrator?

Thanks.

dorodnic, 2019-11-19:
Hi @JBBee 
Sorry for the delay. System time and Global time should be for most part comparable (up to stream latency). There is currently no way to force T265 to system time.
The script needs to be executed once per camera, and there is the alternative of installing [Windows driver](https://downloadcenter.intel.com/download/28723/Intel-RealSense-D400-Series-Universal-Windows-Platform-UWP-Driver-for-Windows-10) once per system.

JBBee, 2019-11-19:
Thanks @dorodnic that answers my question.

closed 2019-11-19 by JBBee