realsenseai/librealsense #4929: Can't build with PCL example.

yokoshou, 2019-09-25:
Hi,my problem is occured when I tried to build with PCL example.

I tried same way of this 
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/pcl
(when I tried,Cmake said"need openCV path".So I choose correct path.)

Solution file generated with no error.
After solution generated,I opened sln with visual studio.
Error occured when I build solution.
Error is "LNK ERROR1104,can't open glfw3.lib"
I tried add glfw3.lib to dependencies and directory,but it caused more many errors.
I already read #3559,but it didn't solve my problem.

Do you have any advice??

-----------------------------------------------
openCV 4.1.0
visual studio 2019 
Cmake 3.15 (build X64)
librealsense newest one
-----------------------------------------------

Marenix, 2019-09-26:
I had a similar issue. I can't remember exactly how I fixed it, but, for me, dependencies of rs-pcl and rs-pcl-color weren't set up right so the projects couldn't find the right files.

Post what errors you're getting after making sure your projects are finding glfw3.lib and I'll try to help you.

yokoshou, 2019-09-30:
Thank you for reply @Marenix.
The errors after add glfw3.lib are
error LNK2019: unresolved external symbol gluPerspective referenced in function "void __cdecl draw_pointcloud(float,float,struct glfw_state &,class rs2::points &)" (?draw_pointcloud@@YAXMMAEAUglfw_state@@AEAVpoints@rs2@@@Z)
1>rs-pcl.obj : error LNK2019: unresolved external symbol gluLookAt referenced in function "void __cdecl draw_pointcloud(float,float,struct glfw_state &,class rs2::points &)" (?draw_pointcloud@@YAXMMAEAUglfw_state@@AEAVpoints@rs2@@@Z)
1>glfw3.lib(input.obj) : error LNK2019: unresolved external symbol _glfwPlatformSetRawMouseMotion referenced in function glfwSetInputMode

...
more than 100 errors.

Could you any advice??

RealSenseCustomerSupport, 2020-01-20:
----------------------------------------------

@yokoshou Sorry for late response! Did you try the latest version to see if you still have this issue? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

@yokoshou Did you get it through now? Looking forward to your reply.Thanks!

RealSenseCustomerSupport, 2020-02-17:
----------------------------------------------

@yokosho Sorry we didn't hear from you for weeks. Will close the ticket at this point. Please feel free to create another ticket if you still have questions or issues. Thanks!

closed 2020-02-17 by RealSenseCustomerSupport