realsenseai/librealsense #4930: How to get the depth value from color frame?

qoo4202w, 2019-09-25:
I used some image processing method to get the position in the color frame by D435i camera. However, how could I get the depth value from that?
I have already checked the example rs-align and rs-align-advanced in IntelRealSense SDK, but I have no idea to find it.

ev-mp, 2019-09-25:
@qoo4202w hello,
In case you are looking for a limited number of points then the `rs2_project_color_pixel_to_depth_pixel` API call should suffice #2948, #4241 
In case you need to extract depth data for multiple features/regions in RGB frame then you you're correct to use `rs-align`.
- Align the depth frame to color, as elaborated in the demo.
- The resulted frameset will include the `aligned_depth` and the original RGB frames.
- For each RGB pixel [i,j] the corresponding depth value would be available at the similar [i,j] index of the `aligned_depth` frame

qoo4202w, 2019-09-29:
@ev-mp 
Thanks for your help. I know how to get depth position from the position of color frame now. 
However, the points given by rs2_project_color_pixel_to_depth_pixel are strange. 
[image]
In the image, the object is only about 30 cm from the camera, but the depth of point is shown as 1.671 meter. It seems straightly sensor the ceiling but not the object.
By the way, I already use the extrinsics and intrinsics parameter from "https://github.com/IntelRealSense/librealsense/wiki/API-How-To".

ravijo, 2019-10-02:
Hi @qoo4202w 

I am not sure, but can you please check the parameters given to `rs2_project_color_pixel_to_depth_pixel`. I was doing a similar experiment in ROS, and I provided wrong pixel values. More specifically, I used the aligned depth image, but the pixels were interchanged. Below is the correct way-

```
auto depth = mDepthImage.at<unsigned short>(static_cast<int>(pixel_y), static_cast<int>(pixel_x));
```
[Link here](https://github.com/ravijo/ros_openpose/blob/b3c4dba1779cb7061f554b6187cfc889eaa73f16/include/ros_openpose/cameraReader.hpp#L88)

RealSenseCustomerSupport, 2019-10-21:
----------------------------------------------

Hi @qoo4202w,

Did @ravijo's answer solve your issue?

Thank you!

RealSenseCustomerSupport, 2019-10-25:
----------------------------------------------

@qoo4202w,

Do you still need help with this question?

closed 2019-10-31 by RealSenseCustomerSupport

BarRae, 2020-04-16:
Hello @qoo4202w 

I don't know if you are going to see this, but I have a question for you.
Could you be so kind as to help me, I am trying to do a similar thing as you did over here.
But I'm having some issue using `rs2_project_color_pixel_to_depth_pixel` and getting the  extrinsics and intrinsics parameters.

Could you maybe provide me with information or a piece of your code?
Sorry if it is all a bit basic what im asking, i am new to opencv and realsense an im trying to learn.

qoo4202w, 2020-04-16:
Hi @BarRae 

I didn't figure out the question in using rs2_project_color_pixel_to_depth_pixel.
However, I solve my problem by using the example "/rs-align-advanced".
https://dev.intelrealsense.com/docs/rs-align-advanced

I modify the example code, and this is my code.

#include <iostream>
#include <opencv2/opencv.hpp>
#include <librealsense2/rs.hpp>
#include <librealsense2/rsutil.h> //Including for frame change

double get_depth_scale(rs2::device dev);
bool profile_changed(const std::vector<rs2::stream_profile>& current, const std::vector<rs2::stream_profile>& prev);

int main() try
{
	// Helper to colorize depth images
	rs2::colorizer color_map;

	// Create a pipeline to easily configure and start the camera
	rs2::pipeline pipe;
	rs2::config pipe_config;

	//adjust resolution for pipelines
	pipe_config.enable_stream(rs2_stream::RS2_STREAM_DEPTH, 640, 480, rs2_format::RS2_FORMAT_Z16, 30);
	pipe_config.enable_stream(rs2_stream::RS2_STREAM_COLOR, 640, 480, rs2_format::RS2_FORMAT_BGR8, 30);
	rs2::pipeline_profile profile = pipe.start(pipe_config);

	//color stream align to depth stream
	rs2_stream align_to = RS2_STREAM_COLOR;
	rs2::align align(align_to);
	double depth_scale = get_depth_scale(profile.get_device());

	//initial all stream
	rs2::frameset camera_framest = pipe.wait_for_frames();
	rs2::frame color = camera_framest.get_color_frame();
	rs2::frame depth = camera_framest.get_depth_frame().apply_filter(color_map);
	rs2::depth_frame frame_d = camera_framest.get_depth_frame();
	rs2::frameset processed = align.process(camera_framest);
	rs2::frame aligned_depth_frame = processed.get_depth_frame().apply_filter(color_map);
	rs2::depth_frame frame_aligned_d = processed.get_depth_frame();

	//get all stream parameter
	const int w_d = depth.as<rs2::video_frame>().get_width();
	const int h_d = depth.as<rs2::video_frame>().get_height();
	const int w_c = color.as<rs2::video_frame>().get_width();
	const int h_c = color.as<rs2::video_frame>().get_height();
	const int depth_w = aligned_depth_frame.as<rs2::video_frame>().get_width();
	const int depth_h = aligned_depth_frame.as<rs2::video_frame>().get_height();

	cv::Mat data_aligned_depth(cv::Size(depth_w, depth_h), CV_16UC1);

	while (cv::waitKey(1) < 0)
	{
		//get frame
		camera_framest = pipe.wait_for_frames();
		color = camera_framest.get_color_frame();
		depth = camera_framest.get_depth_frame().apply_filter(color_map);
		frame_d = camera_framest.get_depth_frame();

		if (profile_changed(pipe.get_active_profile().get_streams(), profile.get_streams()))
		{
			//If the profile was changed, update the align object, and also get the new device's depth scale
			profile = pipe.get_active_profile();
			align = rs2::align(align_to);
			depth_scale = get_depth_scale(profile.get_device());
		}

		processed = align.process(camera_framest);
		aligned_depth_frame = processed.get_depth_frame().apply_filter(color_map);

		frame_aligned_d = processed.get_depth_frame();

		//frame to MAT
		cv::Mat aligned_depth_image(cv::Size(depth_w, depth_h), CV_8UC3, (void*)aligned_depth_frame.get_data(), cv::Mat::AUTO_STEP);
		cv::Mat img_color(cv::Size(w_c, h_c), CV_8UC3, (void*)color.get_data(), cv::Mat::AUTO_STEP);

		//write aligned depth image
		int nr = data_aligned_depth.rows; // number of rows
		int nc = data_aligned_depth.cols * data_aligned_depth.channels(); // total number of elements per line
		for (int j = 0; j < nr; j++) {
			ushort* data = data_aligned_depth.ptr<ushort>(j);
			for (int i = 0; i < nc; i++) {
				//distance*1000 save into ushort(0-65535)
				data[i] = frame_aligned_d.get_distance(i, j) * 1000;
			}
		}

		cv::Mat data_aligned_depth_show(cv::Size(w_d, h_d), CV_8UC1);

		data_aligned_depth.convertTo(data_aligned_depth_show, CV_8UC1, 1, 0);

		cv::imshow("Aligned Depth Frame", aligned_depth_image);
		cv::imshow("Aligned Depth Frame data", data_aligned_depth_show);
		cv::imshow("window_color", img_color);
	}

	return EXIT_SUCCESS;
}
catch (const rs2::error & e)
{
	std::cerr << "RealSense error calling " <<
		e.get_failed_function() << "(" <<
		e.get_failed_args() << "):\n    " <<
		e.what() << std::endl;
	return EXIT_FAILURE;
}
catch (const std::exception & e)
{
	std::cerr << e.what() << std::endl;
	return EXIT_FAILURE;
}

double get_depth_scale(rs2::device dev)
{
	// Go over the device's sensors
	for (rs2::sensor& sensor : dev.query_sensors())
	{
		// Check if the sensor if a depth sensor
		if (rs2::depth_sensor dpt = sensor.as<rs2::depth_sensor>())
		{
			return dpt.get_depth_scale();
		}
	}
	throw std::runtime_error("Device does not have a depth sensor");
}

bool profile_changed(const std::vector<rs2::stream_profile>& current, const std::vector<rs2::stream_profile>& prev)
{
	for (auto&& sp : prev)
	{
		//If previous profile is in current (maybe just added another)
		auto itr = std::find_if(std::begin(current), std::end(current), [&sp](const rs2::stream_profile& current_sp) { return sp.unique_id() == current_sp.unique_id(); });
		if (itr == std::end(current)) //If it previous stream wasn't found in current
		{
			return true;
		}
	}
	return false;
}