realsenseai/librealsense #4931: D435 at Ubuntu 18.04, rs2_pipeline_start_with_config error

mipsan, 2019-09-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400| 
| Firmware Version                   | 5.10.06 / 5.11.11.100
| Operating System & Version |  Linux (Ubuntu 118.04 LTS 
| Kernel Version (Linux Only)    |  4.9.140
| Platform                                 | NVIDIA Jetson Nano
| SDK Version                            |  { legacy / 2.<?>.<?jetson> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Dear,

I installed D435 at Jetson Nano based Kaya as guide.
(https://docs.nvidia.com/isaac/isaac/packages/sensors/doc/realsensecamera.html)

2 things to ask:

1.  realsense-viewer:
It shows both depth and RGB video together, but following message is displayed at startup

> 25/09 12:48:09,378 WARNING [547757199808] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.

And even I use USB 3.x{Jetson Nano has 4 x USB 3.x interfaces on the board), viewer shows USB 2.1 connection. I tested this both 5.10.06 and 5.11.11.100.

2. Kaya, NVIDIA's robot platform, cannot show anything. When it starts, following error messages I met.

> 2019-09-25 12:39:43.385 WARNING packages/realsense/RealsenseCamera.cpp@37: Realsense recommended firmware version is 05.10.03.00, currently using firmware version 05.10.06.00
> 2019-09-25 12:39:44.293 ERROR   engine/alice/component.cpp@61: Component 'camera/realsense' reported an error: RealSense error calling rs2_pipeline_start_with_config(pipe:0x7f84067620, config:0x7f8406a0e0): Couldn't resolve requests
> 2019-09-25 12:39:44.293 ERROR   engine/alice/backend/event_manager.cpp@40: Stopping node 'camera' because it reached status 'FAILURE'
> 2019-09-25 12:39:44.293 DEBUG   engine/alice/backend/codelet_backend.cpp@73: Starting codelet 'camera/realsense' DONE
> 2019-09-25 12:39:44.293 WARNING engine/alice/backend/codelet_backend.cpp@281: Codelet 'camera/realsense' was not added to scheduler because no tick method is specified.
> 

Application to test is here: https://docs.nvidia.com/isaac/isaac/doc/tutorials/assemble_kaya.html#the-object-detection-application

Any idea please? 
Thanks,

ev-mp, 2019-09-25:
@mipsan  hello,
The specific error appear when the application tries to configure stream with unsupported resolution/fps/format modes. When device is recognized as USB2 it only provides a subset of resolutions/fps modes sustainable with the reduced bandwidth and power budgets.
Run `rs-enumerate-devices -o` to list all the supported configurations, and then reconfigure the Librealsense code accordingly.

As for the difference in `dev/video*` - this is normal. With kernels 4.16+ for each video device two media nodes are created (video+metadata)

mipsan, 2019-10-02:
> @mipsan hello,
> The specific error appear when the application tries to configure stream with unsupported resolution/fps/format modes. When device is recognized as USB2 it only provides a subset of [...]

@ev-mp ,

Thank you very much.

closed 2019-10-02 by mipsan