realsenseai/librealsense #4932: rs-rosbag-inspector not working

pncanas333, 2019-09-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Operating System & Version |   {Win (10) / Linux (Ubuntu 16) | 

Hello!

I have trouble initializing the rs-rosbag-inspector tool. It seems to be a problem with the latest releases since it worked perfectly a couple of weeks ago (don't remember what version I had). I installed the latest version of the SDK (2.28.1) on both Windows and Ubuntu and it doesn't work now. 

In Ubuntu, it shows this error: Segmentation fault (core dumped).

Thanks.

RealSenseCustomerSupport, 2019-10-15:
----------------------------------------------

Hi @pncanas333,

Please list all the platforms (board, OS and build/SP/kernel version on this board) you have segmentation fault on and specify Intel RealSense SDK version installed on each of these platforms.
Do you use ROS Bag Inspector that is coming as a part of the Intel RealSense SDK you have installed or you use other version(please specify what version)?
Along with that information, please describe the exact scenario that leads to the problem.

Thank you!

RealSenseCustomerSupport, 2019-10-22:
----------------------------------------------

Hi @pncanas333,

Do you still need help with this question?

closed 2019-10-29 by RealSenseCustomerSupport