realsenseai/librealsense #4937: Disable of device-initiated U1/U2 failed

norman1479, 2019-09-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435i | 
| Firmware Version                   | 5.11.10.0 | 
| Operating System & Version |   Linux Ubuntu 18.04  | 
| Kernel Version (Linux Only)    | 4.15.0-55 or 58                                         | 
| Platform                                 | Intel i7 Q370 PC |
| SDK Version                            | 2.28.1                          | 
| Language                            |  C++                          | 
| Segment			| Service Robot                           | 

### Issue Description

We have multiple (up to 4) D435i connected to the PC's USB3.1 Gen1 ports. During synchronous startup or longer operation using realsense-ros (or sometimes even with realsense-viewer), unexpected resets occur according to the syslog:
`kernel: [ 1043.112609] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024)`
`kernel: [ 1043.297456] usb 2-4: Disable of device-initiated U1 failed.`

`kernel: [ 1043.304451] usb 2-4: Disable of device-initiated U2 failed.`

`kernel: [ 1043.418840] usb 2-4: reset SuperSpeed Gen 1 USB device number 3 using xhci_hcd`

`kernel: [ 1043.525437] usb 2-7: Disable of device-initiated U1 failed.`

` kernel: [ 1043.532429] usb 2-7: Disable of device-initiated U2 failed.`

`kernel: [ 1043.646688] usb 2-7: reset SuperSpeed Gen 1 USB  device number 4 using xhci_hcd`

`kernel: [ 1046.164230] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).`

Adding delays between initial resets, startup or disabling USB autosuspend/auto-poweroff in the kernel/udev only seemed to have reduced the frequency of such events in our setup, but they still occur and make librealsense/realsense-ros crash at xioctl calls.

We also tested the cameras at the PC's USB3.1 Gen2 ports instead of Gen1, but these tended to operate much worse in our case.

Any solutions or hints on how to effectively avoid this?

Thanks in advance and best regards
Norman

RealSenseCustomerSupport, 2019-10-14:
----------------------------------------------

> During synchronous startup

Are the cameras used in multi-camera hardware sync configuration? The problem sounds like it's related to ESD/EMI related.
The multi-camera white paper provides good recommendations to ensure the signal integrity and protects the cameras.
https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration
Thank you.

closed 2019-10-23 by RealSenseCustomerSupport

hugeturing, 2022-08-16:
Same issue without hardware sync. Please help!
`input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4.3/2-4.3:1.0/input/input3147
usb 2-4.3: Disable of device-initiated U1 failed.
usb 2-4.3: Disable of device-initiated U2 failed.
usb 2-4.3: reset SuperSpeed Gen 1 USB device number 51 using xhci_hcd
uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 410  (8086:0ad2)
uvcvideo 2-4.3:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
uvcvideo 2-4.3:1.0: Entity type for entity Processing 2 was not initialized!
uvcvideo 2-4.3:1.0: Entity type for entity Camera 1 was not initialized!
input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4.3/2-4.3:1.0/input/input3148
usb 2-4.3: Disable of device-initiated U1 failed.
usb 2-4.3: Disable of device-initiated U2 failed.
usb 2-4.3: reset SuperSpeed Gen 1 USB device number 51 using xhci_hcd
uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 410  (8086:0ad2)
uvcvideo 2-4.3:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
uvcvideo 2-4.3:1.0: Entity type for entity Processing 2 was not initialized!
uvcvideo 2-4.3:1.0: Entity type for entity Camera 1 was not initialized!`

milidam, 2024-11-15:
Hi,

I am encountering the same kind of behavior with a D455 camera.
The camera has its own power supply, and hardware sync is not configured.

Any idea what could cause it? and how to fix or workaround it?

MartyG-RealSense, 2024-11-16:
Hi @milidam  I recommend trying to build the librealsense SDK from source code with the **libuvc backend** installation method, which bypasses the kernel and so can help to avoid problems that are arising from a conflict between librealsense and the kernel.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh