realsenseai/librealsense #4938: Wheel odometry to T265 with two wheels

mzelina, 2019-09-25:
I have a two wheel robot and want to feed odometry data into the T265 to aid in tracking. I have read the documentation and surrounding articles (e.g. https://github.com/IntelRealSense/librealsense/pull/3462#issuecomment-472575654, https://github.com/IntelRealSense/librealsense/issues/3970#issuecomment-492111666).

The JSON file example describes T as "relative transformation of sensor frame w.r.t. T265 body frame (see T265 data sheet, sec. 4.3 "Coordinate System"). I am not sure if this is the the center of the wheel, where the wheel contacts the surface or some other measurement? 

Similar question for W described as "the orientation in axis-angle representation, axis x angle (in rad)".  I don't understand how this axis-angle representation maps to direction. One wheel is going positive and one negative to move forward. Is this where this transformation comes into account?

My wheels are mounted behind the camera with the camera centered on the chasis and the wheels being ~0.7m behind and ~0.1m below the center of the camera and a wheel base of 0.46m.

My best guess is these items in my file would be the following for translations (intentionally ommited the other parts of the file). For W I am not sure. Any help appreciated!

"velocimeters": [
        { // Wheel left
            ...
            "extrinsics": {
                "T": [
                    -0.23,
                    0.7,
                    -0.1
                ],
                "W": [ // orientation in axis-angle representation, axis x angle (in rad)
                    ????,
                    ????,
                    ????
                ]
            }
        },
      { // Wheel right
            ...
            "extrinsics": {
                "T": [
                    0.23,
                    0.7,
                    -0.1
                ],
                "W": [ // orientation in axis-angle representation, axis x angle (in rad)
                    ????,
                    ????,
                    ????
                ]
            }
        }
    ]
}

mzelina, 2019-09-28:
Missed this previously but the unit test here is helpful if anyone else has issues:

https://github.com/IntelRealSense/librealsense/blob/master/unit-tests/unit-tests-live.cpp#L5535

Still not clear on how exactly to calculate the axis-angle from the robot geometry.

mzelina, 2019-10-01:
@dorodnic Can you clarify something in your test case for the wheel odometry?

https://github.com/IntelRealSense/librealsense/blob/master/unit-tests/unit-tests-live.cpp#L5535

You are expecting the robot to be stationary in your comments but you are passing 1, 0, 0 for the translational velocity vector. I would have expected that to be 0, 0, 0.

Also, are there any real-world examples for odometry calibration? Perhaps the odometry calibration for the kobuki tests you ran would be useful.

schmidtp1, 2019-10-15:
@mzelina,
> I have a two wheel robot and want to feed odometry data into the T265 to aid in tracking. I have read the documentation and surrounding articles (e.g. [#3462 (comment)](https://github.com/IntelRealSense/librealsense/pull/3462#issuecomment-472575654), [#3970 (comment)](https://github.com/IntelRealSense/librealsense/issues/3970#issuecomment-492111666)).
> 
> The JSON file example describes T as "relative transformation of sensor frame w.r.t. T265 body frame (see T265 data sheet, sec. 4.3 "Coordinate System"). I am not sure if this is the the center of the wheel, where the wheel contacts the surface or some other measurement?
This is where the wheel contacts the surface.
> 
> Similar question for W described as "the orientation in axis-angle representation, axis x angle (in rad)". I don't understand how this axis-angle representation maps to direction. One wheel is going positive and one negative to move forward. Is this where this transformation comes into account?
There is some design freedom in the definition of the velocity frame. I usually align the x-axis with the velocity vector and then define the rotation between the velocity frame and the T265 in the calibration file.
Yes, it can be used to account for that difference between the two wheels.
>
> My wheels are mounted behind the camera with the camera centered on the chasis and the wheels being ~0.7m behind and ~0.1m below the center of the camera and a wheel base of 0.46m.
> 
> My best guess is these items in my file would be the following for translations (intentionally ommited the other parts of the file). For W I am not sure. Any help appreciated!
> 
> ```
> "velocimeters": [
>         { // Wheel left
>             ...
>             "extrinsics": {
>                 "T": [
>                     -0.23,
>                     -0.1,
>                     0.7
>                 ],
>                 "W": [ // orientation in axis-angle representation, axis x angle (in rad)
>                     -1.2092,
>                     -1.2092,
>                     -1.2092
>                 ]
>             }
>         },
>       { // Wheel right
>             ...
>             "extrinsics": {
>                 "T": [
>                     0.23,
>                     -0.1,
>                     0.7
>                 ],
>                 "W": [ // orientation in axis-angle representation, axis x angle (in rad)
>                     -1.2092,
>                     1.2092,
>                     1.2092
>                 ]
>             }
>         }
>     ]
> }
> ```
The translation is correct for the case that you describe.
I filled in the orientation according to the small sketch below. (Please feel free to double-check it.)
[image]

RealSenseCustomerSupport, 2019-10-22:
----------------------------------------------

Hi @mzelina - Were you able to use the information provided by schmidtp1?  Anything else needed?

closed 2019-11-14 by RealSenseCustomerSupport